ArduPilot/MissionPlanner #2263: System.MissingMethodException(MAVLinkInterface.add_UpdateADSBCollision)

Chatmetaleux, 2019-11-28:
#### Issue details
Crash at startup with method not found in Program.cs :
System.MissingMethodException: Méthode introuvable : 'Void MissionPlanner.MAVLin
kInterface.add_UpdateADSBCollision(System.EventHandler`1<System.ValueTuple`2<Sys
tem.String,MAV_COLLISION_THREAT_LEVEL>>)'

Works well with 1.3.68 (Win7)

#### Version
1.3.69 build 1.3.7271.14198

#### Platform
na

#### Airframe type
na

#### Hardware type
na

#### Logs
Debug under mono    MONO_LOG_LEVEL=debug mono MissionPlanner.exe
To fix any filename case issues under mono use    export MONO_IOMAP=drive:case
Data Dir C:\ProgramData\Mission Planner\
Log Dir C:\Users\Drone\Documents\Mission Planner\logs
Running Dir C:\Program Files (x86)\Mission Planner\
User Data Dir C:\Users\Drone\Documents\Mission Planner\
INFO Program - ******************* Logging Configured *******************
INFO Program - SkiaLoaded
INFO Program - 64bit os True, 64bit process True
FATAL Program - Fatal app exception
System.MissingMethodException: Méthode introuvable : 'Void MissionPlanner.MAVLin
kInterface.add_UpdateADSBCollision(System.EventHandler`1<System.ValueTuple`2<Sys
tem.String,MAV_COLLISION_THREAT_LEVEL>>)'.
   à MissionPlanner.MainV2..ctor()
   à MissionPlanner.Program.Start(String[] args) dans C:\Users\michael\Source\Re
pos\MissionPlanner\Program.cs:ligne 263
System.MissingMethodException: Méthode introuvable : 'Void MissionPlanner.MAVLin
kInterface.add_UpdateADSBCollision(System.EventHandler`1<System.ValueTuple`2<Sys
tem.String,MAV_COLLISION_THREAT_LEVEL>>)'.
   à MissionPlanner.MainV2..ctor()
   à MissionPlanner.Program.Start(String[] args) dans C:\Users\michael\Source\Re
pos\MissionPlanner\Program.cs:ligne 263

Press any key to exit!

meee1, 2019-11-28:
please reinstall. for some reason one of the dll's must have failed to update.

Chatmetaleux, 2019-11-29:
It worked after 2 tries...
Thanks

closed 2019-11-29 by Chatmetaleux