ArduPilot/MissionPlanner #2264: InitialSetup/Compass: Remove "Enable compasses" checkbox

rmackay9, 2019-11-29:
We've removed the MAG_ENABLE parameter from recent builds and it's a real edge condition that someone would want to disable compasses.  It's such an edge case that we can remove it from this screen entirely I think because it just causes confusion.

[image]

meee1, 2019-11-29:
i will hide it if the param doesnt exist. else it will break older users

meee1, 2019-11-29:
or i could just remove it?

rmackay9, 2019-11-29:
@meee1, I think it can just be removed completely regardless of which version is being used.

I think the only place where a user would want to disable the compasses is when doing non-GPS navigation and the wiki setup guides specify to set the MAG_ENABLE parameter to zero ([like here](http://ardupilot.org/copter/docs/common-vio-tracking-camera.html#configure-ardupilot)) so we don't need any easy-to-use checkbox for the 99% of users who are not doing non-GPS navigation