ArduPilot/MissionPlanner #2265: Flight Plan map doesn't recenter on vehicle

rmackay9, 2019-12-03:
I've noticed two unrelated issues with the Flight Plan map:

- there is no way to easily recenter the map on the vehicle which means that the user must manually zoom out, center the red plus on the vehicle and zoom back in again.  I think it should perhaps have an "Auto Pan" checkbox like the main Flight Data screen.
- the map jitters up and down if the red plus is placed over the green Home marker and the map is then moved up and down by left-mouse-button-click on the bottom right of the map (see red X on image below)
[image]

meee1, 2019-12-04:
1. by default it zooms to your home location. to make it jump to the drone, click "home location"
2. i will need to look further

meee1, 2019-12-05:
2 is because you are trying to pan the "center" off the edge of the screen, so it jumps around.  ie the top of the map cant go past 50% of the screen

rmackay9, 2019-12-05:
hm... not sure, but it's very odd looking.  I'll leave it with you though, feel free to close

Naterater, 2019-12-06:
Why don't we make the home location default to the vehicle's position UNLESS the user has set it. Wouldn't that be a fix for this problem plus the reset to 0,0 Lat/Long when doing things like clearing a mission?

rmackay9, 2019-12-14:
I tried clicking on "Home Location" while the Flight Plan map was centered way off in Europe somewhere.. the map didn't re-center on the vehicle (which is in Japan).. maybe I'm misunderstanding but I can't figure out a way to easily make the Flight Plan map recenter on the vehicle.. I always need to zoom out, center the red cross on the vehicle zoom in, re-center, zoom in, etc.

rmackay9, 2020-01-31:
Re the first issue, clicking on the "Home Location" on the Flight Plan screen does not recenter the Flight Plan screen's map.  Below is a screen shot of what I see on the Plan screen:
[image]
Clicking on the Home Location does update the Lat Lon fields but it's not re-centering the map.