ArduPilot/MissionPlanner #2270: MAVFtp broken in 1.3.70, working in 1.3.69

olliw42, 2019-12-07:
#### Issue details
With my setup, MAVlinkFTP using the cntrl-X command is working great in version 1.3.69, but not in version 1.3.70. In version 1.3.70 the MAVFtp window opens, but doesn't show any files, or do anything. To me it looks as if the MAVTfp listdir cmd isn't working anymore. This is very reproducible, I've reinstalled v1.3.69 and v1.3.70 several times, and each time it is working with v1.3.69 but not with v1.3.70.

The setup is a arducopter flight controller with a storm32 gimbal controller, and the MAVLinkFTP is with the storm32 component. 

I went through the commits since v1.3.69 but couldn't find any which I could relate to MAVLinkFTP and change in its behavior. I saw some changes in MavlinkInterface, e.g. this await thing, so one may speculate it's somehow related to this, but I certainly have no idea.

#### Version
v1.3.70, and latest

#### Platform
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
irrelevant

#### Hardware type
fmuv3, but is irrelevant 

#### Logs
[mp1370.zip](https://github.com/ArduPilot/MissionPlanner/files/3935114/mp1370.zip)

meee1, 2019-12-07:
I've fixed this in the latest beta

olliw42, 2019-12-08:
indeed working again. GREAT :)
close it.

closed 2019-12-08 by olliw42