ArduPilot/MissionPlanner #2278: Add desired Alt Refference MSL or AGL to ConfigPlanner

jfint, 2019-12-11:
#### Issue details
Many times I want to view HUD altitude in MSL values not AGL.  I can do this with a simple local edit in CurrentState.cs, but it would be much better if this were a drop-down or check-box in the planner configuration page like the altitude units.

It looks like this changed in 2013

#### Version
1.3.70

#### Platform
[ X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
 Quad, Flying Wing, Standard Fixed Wing)

#### Hardware type
FMUv2 and FMUv5

#### Logs
N/A

meee1, 2019-12-11:
In action tab, click set home alt

jfint, 2019-12-11:
Will this still work if your home alt happens to not be on the ground?  It is looking like I'm batting 0 for today ;-)  Thanks for your help!

meee1, 2019-12-11:
im pretty sure it will work as you expect it to

closed 2019-12-11 by jfint