ArduPilot/MissionPlanner #2279: Feature request: Add LTM support

vierfuffzig, 2019-12-12:
LTM support has been added to Ardupilot firmware recently. It‘s an ultimately lightweight telemetry protocol that allows to downlink the most vital flight data (position, attitude, status, navigation) at 1200 / 2400 baud while still providing update rates of 5-10 Hz.
Due to its design it is particularly suitable for long range downlink telemetry over AV audio, using a simple FSK modem, without the need for additional radio / HF components.
On the ground side, it is frequently used for antenna tracking or simple ground-based data display (Ground-OSD). 
However, the list of GCSes that natively support LTM is somewhat limited, actually there’s only one linux-based app still being actively developed (mwptools) as to my knowledge (https://github.com/stronnag/mwptools).

Adding LTM support to MissionPlanner would allow for seamless integration of this streamlined telemetry protocol into the ardupilot environment.

meee1, 2019-12-12:
where is the information about the protocol itself. is it two way?

vierfuffzig, 2019-12-12:
thanks @meee1 for looking into this!

summary:
https://quadmeup.com/ltm-light-telemetry-protocol/

original code by KipK:
https://github.com/artemen/Ghettostation/blob/master/GhettoProxy/LightTelemetry.cpp

ardupilot AP_LTM:
https://github.com/ArduPilot/ardupilot/pull/12158/files

forum thread:
https://discuss.ardupilot.org/t/adding-ltm-telemetry-support/35385

afaik it‘s been genuinely designed for an ultimately lean unidirectional data downlink.

mariansoban, 2019-12-12:
> 
> 
> where is the information about the protocol itself. is it two way?

Actually it's one-way protocol.