ArduPilot/MissionPlanner #2282: Stop screwing with .bin file extension

hiseob, 2019-12-20:
#### Issue details

Mission Planner installer screws with .bin file extension and it bring pains in ass

I don't know how this happen but somehow mission planner installer screwed with .bin extension, binary files from Keil MDK are gone undeletable unless reboot into safe-mode, actually explorer.exe loads the bin as dll then file locks up.

anyway ".BIN" represents BINary not "ArduPilot Log File"

If you want run mission planner with double click on file, please fix the firmware first (introduce new file extension) , then associate this new file extension

many user/devs are not using mission planner alone in windows

.bin associated with mission planner and the MP icon ... absolutly disgusting every time i see it 

#### Version

all versions i've used few years

jmachuca77, 2019-12-20:
You can change the association back to whatever it was before: 
https://www.google.com/amp/s/www.thewindowsclub.com/change-file-associations-windows/amp

hiseob, 2019-12-20:
> You can change the association back to whatever it was before:
> https://www.google.com/amp/s/www.thewindowsclub.com/change-file-associations-windows/amp

funny :)

proficnc, 2019-12-20:
The only way .bin becomes associated with mission planner is if you tell it too. 

The .bin file is a binary file, it is correctly used as a binary file. 

Follow the directions that Jaime gave you to change any association you have added.

The .bin file is an ardupilot thing, Mission Planner simply uses the file.

hiseob, 2019-12-20:
> The only way .bin becomes associated with mission planner is if you tell it too.
>
> The .bin file is a binary file, it is correctly used as a binary file.
>
> Follow the directions that Jaime gave you to [...]

mission planner installer associates automatically without notification/confirmation and this is not good behaviour because the .bin extension is used very commonly in development tools , ides etc.

this is why i raising issue , not because I don't know how deleting association registry with regedit.exe

run regedit.exe and find .bin with whole word option and delete registries aren't that hard, already i know