ArduPilot/MissionPlanner #2283: new serial drivers break F7 boards

tridge, 2019-12-25:
There seems to have been an update to the USB drivers recently which has broken F7 boards very badly. 
For example, if a user has a CUAVv5Nano and they set SERIAL7_PROTOCOl=22 to enable SLCAN on the 2nd USB endpoint then only one USB endpoint appears in windows, and that one endpoint is the SLCAN one. This means the user has no way to fix the issue except by firing up Linux so that can access both endpoints.
The same board shows up correctly as two endpoints in Linux.

meee1, 2019-12-25:
How does the device identify on the bus? 
A vid and PID can only have one definition. Ie either single CDC ACM, or dual CDC ACM. Not changing between both. If you want to change between both change the vid/pid

tridge, 2019-12-25:
> How does the device identify on the bus?

the same as CubeOrange, two endpoints

> A vid and PID can only have one definition. Ie either single CDC ACM, or dual CDC ACM. Not changing between both. If you want to change between both change the vid/pid

hmm, then we'll need a new VID/PID for these devices.

I suggest we use VID 0x1209 PID 0x5741. This is allocated to ArduPilot here:

https://github.com/pidcodes/pidcodes.github.com/blob/master/1209/5740/index.md

I'll try to put together a inf file based on what you did for CubeOrange

Also, I have found a workaround for the issue so user can "unbrick". If you manually change the driver in windows 10 to "USB Composite Device" then you can access the two endpoints separately.

tridge, 2019-12-25:
given this limitation, how does CubeOrange work both in the bootloader and normal? When it is in the bootloader it only has one endpoint. When it is in normal operation it has two, but in both cases it works in windows and MissionPlanner can talk to it.

tridge, 2019-12-25:
I've opened a PR for the PID change:
https://github.com/ArduPilot/ardupilot/pull/13115
this also will add the USB VID/PID to the apj file and use that to populate the IDs in the manifest

tridge, 2019-12-25:
I've opened a PR for mission planner to try to fix this:
https://github.com/ArduPilot/MissionPlanner/pull/2284
it assigns the USB IDs as follows:

single endpoint: 0x0483/0x5740
single endpoint: 0x1209/0x5741
dual endpoint: 0x1209/0x5740

we aren't using 0x1209/0x5741 yet, but I'd like it there to be ready for future use

closed 2022-07-20 by tridge