ArduPilot/MissionPlanner #2285: Complex fence support issues (affecting Copter and Rover)

rmackay9, 2020-01-04:
This is a list of known issues with the MP's support of Complex Fence support that was introduced with ArduPilot Copter/Rover 4.0.

- "There was an unexpected error" message when attempting to upload a polygon fence.  This was a very simple fence as shown below:
[image]

meee1, 2020-01-15:
I believe this is resolved now as well

rmackay9, 2020-01-27:
Issues uploading fences to Copter-4.0 came up today in the support forums here: https://discuss.ardupilot.org/t/impossible-to-upload-the-new-geofence-in-copter-4-0-1-in-holybro-pixhawk-4/51589

rmackay9, 2020-01-27:
@meee1,

I retested this today and it's definitely gotten better thanks.

I immediately bumped into the first reported problem, ("There was an unexpected error") because I did not realise what I needed to put into the "Points" fields for a polygon fence.  I guess the user has to type in the number of points in the polygon that they are currently adding.  So for example if I had an 8 point polygon, I'd first click on all the points and then type "8" into all the Points fields.  It's kind of OK for users who are used to spreadsheets, for regular people I fear it's a bit uintuitive and they will end up fiddling around with the "Points" columns.  I'm also wonder what the polygon drawer is for now..

Kwapin, 2020-06-24:
Has Anybody resolved the issue number 1? I have the same error and cannot upload fences to Ardurover 4.0 over SiK radio.

rmackay9, 2020-06-24:
@Kwapin,
I think issue 1 is resolved now.  We've got a wiki page here which tries to describe it: https://ardupilot.org/copter/docs/common-polygon_fence.html.

I think that MP's interface is a little difficult to understand still but this is the process:

- Go to the Pl;an screen
- Select Fence from the top right corner of the map
- Select the polygon icon on the far left side of the screen
- Draw a polygon on the map by clicking
- Select the polygon icon *again* on the far left side of the screen and select Fence Inclusion or Fence Exclusion

Hope that helps

meee1, 2020-06-24:
@Kwapin make sure you enable mavlink2

Kwapin, 2020-06-24:
> @Kwapin make sure you enable mavlink2

That was it, thanks @meee1 seems like enabling mavlink2 solved the issue,
@rmackay9 Thanks for help, that is all clear.
Seems it would be good idea to add to wiki information that this bug comes from wrong mavlink setting.

rmackay9, 2020-06-25:
@Kwapin,

Ah, right.  I actually didn't realise this was a requirement.  [Issue here for the wiki](https://github.com/ArduPilot/ardupilot_wiki/issues/2874) and I think we will default the protocol to Mavlink2 for ArduPilot-4.1 (Copter, Plane, Rover, etc)

Hwurzburg, 2020-06-25:
When do you need Mav2? fence upload/download only? Plane geofences (they use a different method, I think)? need some specifics if we are going to add to the wiki

rmackay9, 2020-06-25:
@Hwurzburg, I'm not sure but I suspect that to upload or download complex fences (polygon or circle) it requires MAVLink2 to be enabled.