ArduPilot/MissionPlanner #2288: auto mode dynamic change speed and alt change doesn't work

kalaiselvan1219, 2020-01-04:
#### Issue details
if copter execute the mission in auto mode with predefined alt and speed ,and i want to change the ground speed and altitude of the model with help of change speed and change alt in action tap of flight data page .but it work with arduplane but not with arducopter.

in auto mode dynamic loiter mode change will cause decent to zero altitude level not maintain at current height

#### Version
mission planner 1.3.70

#### Platform
[  ] mission planner

#### Airframe type
_Arducopter 3.6.11 HEXA

#### Hardware type
_CUAV V5+

#### Logs
_Please provide a link to any relevant logs that show the issue_

rmackay9, 2020-01-06:
It would be best to raise this in the Copter issues list because it's a flight code issue and not a mission planner issue. https://github.com/ArduPilot/ardupilot/issues.

I think I'll close this if that's OK.

closed 2020-01-06 by rmackay9

kalaiselvan1219, 2020-01-06:
Thank you. I will post it in arducopter page. Will it be consider for
further firmware updates?

rmackay9, 2020-01-06:
@kalaiselvan1219, thanks.  Yes, the issue with loitering in Auto seems like a bug so we should certainly fix that one if we can reproduce it. The change in speed may already be fixed in 4.0.0 but in any case we should make that work.  I'm less sure about how we would handle the change-alt command but it's still a valid request.

kalaiselvan1219, 2020-01-07:
I have one doubt randy.
If  arducopter flying in auto flight mode and if suddenly we have activated
GPS denied environment (GPS will not work). What will happen to the model.
Whether its follow with current heading or will it wait(hover) for gps
signal to regain.