ArduPilot/MissionPlanner #2289: 1.3.7277.34800 Fatal app exception on startup

teodoryantcheff, 2020-01-05:
#### Issue details
Fatal app exception on startup after upgrade to latest MP.

#### Version
1.3.70 build 1.3.7277.34800

#### Platform
[X] All

```D:\Mission Planner>MissionPlanner.exe
If your error is about Microsoft.DirectX.DirectInput, please install the latest directx redist from here http://www.microsoft.com/en-us/download/details.aspx?id=35

Debug under mono    MONO_LOG_LEVEL=debug mono MissionPlanner.exe
To fix any filename case issues under mono use    export MONO_IOMAP=drive:case
Data Dir C:\ProgramData\Mission Planner\
Log Dir C:\Users\xxxx\Documents\Mission Planner\logs
Running Dir D:\Mission Planner\
User Data Dir C:\Users\xxxx\Documents\Mission Planner\
INFO Program - ******************* Logging Configured *******************
INFO Program - SkiaLoaded
INFO Program - 64bit os True, 64bit process True
FATAL Program - Fatal app exception
System.MissingMethodException: Method not found: 'Void MissionPlanner.MAVLinkInterface.add_UpdateADSBCollision(System.EventHandler`1<System.ValueTuple
`2<System.String,MAV_COLLISION_THREAT_LEVEL>>)'.
   at MissionPlanner.MainV2..ctor()
   at MissionPlanner.Program.Start(String[] args)
System.MissingMethodException: Method not found: 'Void MissionPlanner.MAVLinkInterface.add_UpdateADSBCollision(System.EventHandler`1<System.ValueTuple
`2<System.String,MAV_COLLISION_THREAT_LEVEL>>)'.
   at MissionPlanner.MainV2..ctor()
   at MissionPlanner.Program.Start(String[] args)

Press any key to exit!

mainv2_Dispose

D:\Mission Planner>
```

`C:\Users\xxxx\Documents\Mission Planner\logs` is empty.

Zipped Log file from ProgramData
[MissionPlanner.zip](https://github.com/ArduPilot/MissionPlanner/files/4023587/MissionPlanner.zip)

Looking at the log (in ProgramData) I think it gets confused as to where the `ProgramData` is given it's installed **not** no the `C` drive.

teodoryantcheff, 2020-01-05:
[MissionPlanner.zip](https://github.com/ArduPilot/MissionPlanner/files/4023595/MissionPlanner.zip)

Log after deleting all from the previous install dir, `C:\ProgramData\Mission Planner`, and `C:\Users\xxxx\Documents\Mission Planner` then doing a clean install from the latest msi.

.net framework 4.7.1 - 4.7.02558
Windows 7x64 Ultimate, core i7, 16 gigs of ram. 
System and MP have been working for years...

meee1, 2020-01-10:
the cause is a bad update. one of the dll's didnt upgrade at a guess. reinstalling should resolve all this

teodoryantcheff, 2020-01-10:
Will try (again) and report here. 

Thanks for your time, Michael!

dobje, 2020-01-10:
Same issue , tried to re-install did not work , removed and installed again did not work , installed 1.3.69 same error again.

teodoryantcheff, 2020-01-10:
```2020-01-10 20:16:38,516  INFO Program - ******************* Logging Configured ******************* (:0) [1]
2020-01-10 20:16:38,594  INFO Program - SkiaLoaded (:0) [1]
2020-01-10 20:16:38,703 DEBUG Program - Loaded: GMap.NET.WindowsForms, Version=0.0.0.0, Culture=neutral, PublicKeyToken=null (:0) [1]
2020-01-10 20:16:38,718 DEBUG Program - TypeResolve Failed: Mono.Runtime from MissionPlanner.Utilities, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null (:0) [1]
2020-01-10 20:16:38,843  INFO Program - 64bit os True, 64bit process True (:0) [1]
2020-01-10 20:16:38,874 DEBUG Program - Loaded: System.Reactive, Version=4.0.0.0, Culture=neutral, PublicKeyToken=94bc3704cddfc263 (:0) [Base Thread]
2020-01-10 20:16:38,890 DEBUG Program - Loaded: ProjNET, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null (:0) [Base Thread]
2020-01-10 20:16:38,890 DEBUG Program - Loaded: GeoAPI.CoordinateSystems, Version=1.7.5.0, Culture=neutral, PublicKeyToken=a1a0da7def465678 (:0) [Base Thread]
2020-01-10 20:16:38,890 DEBUG Program - TypeResolve Failed: Mono.Runtime from MissionPlanner.ArduPilot, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null (:0) [Base Thread]
2020-01-10 20:16:38,906 DEBUG Program - Loaded: System.ValueTuple, Version=4.0.2.0, Culture=neutral, PublicKeyToken=cc7b13ffcd2ddd51 (:0) [Base Thread]
2020-01-10 20:16:38,921 DEBUG Program - FirstChanceException in: MissionPlanner (:0) [Base Thread]
System.MissingMethodException: Method not found: 'Void MissionPlanner.MAVLinkInterface.add_UpdateADSBCollision(System.EventHandler`1<System.ValueTuple`2<System.String,MAV_COLLISION_THREAT_LEVEL>>)'.
   at MissionPlanner.MainV2..ctor()
   at MissionPlanner.Program.Start(String[] args)
2020-01-10 20:16:38,921 FATAL Program - Fatal app exception (:0) [Base Thread]
System.MissingMethodException: Method not found: 'Void MissionPlanner.MAVLinkInterface.add_UpdateADSBCollision(System.EventHandler`1<System.ValueTuple`2<System.String,MAV_COLLISION_THREAT_LEVEL>>)'.
   at MissionPlanner.MainV2..ctor()
   at MissionPlanner.Program.Start(String[] args)
```

meee1, 2020-01-10:
check here
https://firmware.ardupilot.org/Tools/MissionPlanner/archive/
are you sure all files are being replaced?
mainly missionplanner.ardupilot.dll?

teodoryantcheff, 2020-01-11:
> are you sure all files are being replaced?

Yes, got the zip and extracted it in a new folder.

teodoryantcheff, 2020-01-11:
Ha, even with the 1.3.69 zip, extracted to `c:\mp` I still get the `Method not found` exception.

meee1, 2020-01-11:
I'm starting to think there is something wrong with your windows install

teodoryantcheff, 2020-01-11:
And I'm not saying you are not wrong... :) 
But that started only after updating to 1.3.70. And everything else is working just as good as it has been for the last 10 (or more) years since windows was installed on the machine. I'll see to digging more into this and report back.

meee1, 2020-01-11:
depending on how technical you want to get, you can use a app like procmon to monitor what missionplanner is doing on startup. see if its looking in the wrong place for the file it needs to load.

could also be some random virus scanner issue blocking access

dobje, 2020-01-11:
> Ha, even with the 1.3.69 zip, extracted to `c:\mp` I still get the `Method not found` exception.

Try 1.3.68

teodoryantcheff, 2020-01-12:
1.3.68 from the archives zip **works** (starts) flawlessly, regardless of which drive I put it on. Thanks @dobje 
Updating with the "built-in" updater -- yields `method not found`. Just like running 1.3.70 from either msi or zip.

teodoryantcheff, 2020-01-13:
Well, after 30ish more minutes looking into this, I assumed it's a .net issue, downloaded 4.8 from https://dotnet.microsoft.com/download/dotnet-framework/thank-you/net48-web-installer and that seems to have resolved the problem.
I did make a few process logs with procmon but lost them in the joy of having a running mission planner. Now both 1.3.69 and 1.3.70 start.

Thank you to both @meee1 and @dobje !

closed 2020-01-13 by teodoryantcheff