ArduPilot/MissionPlanner #2291: Compatibility with dot core

khancyr, 2020-01-06:
#### Issue details
It seems that Microsoft is betting on dot core over dot framework, so preparing the migration would be a good things.
Moreover this will bring some good improvement like a better cross platform support and docker support. With some hack both are already working but far from stable and out of reach of inexperienced user.

### Compatibility file
In order to help us, I have the compatiblity tool on current version.

here is the result file (a simple html file in a zip to be able to put it in github) : 
[ApiPortAnalysis.zip](https://github.com/ArduPilot/MissionPlanner/files/4026356/ApiPortAnalysis.zip)

davidbuzz, 2020-01-13:
Microsoft opensourced the non-graphical bits of .net calling it '.net core' , so theres currently no way to build a classic/windows .net app that uses any on-screen or gui bits, using the open sourced 'core'. Core is ok for servers, backend's, networking, etc... and a little bit of research shows that MP has been progressing toward more use of 'core' over time, but my understanding is that 'core' is never going to completely replace classic .net in the GUI space, or at least not for the forseeable future.

jverhuls, 2021-01-19:
I agree that multi-OS support would be cool! But indeed the .net core doesn't provide currently a good gui for multplatform.
2 options: Use a framework like Avalonia or wait for for the upcomming .net MAUI framework.

Would a move to Avalonia be possible?

meee1, 2021-01-19:
currently i have the current master running on android, osx, ios, windows, linux via either mono/ dotnet or xamarin

meee1, 2021-01-19:
check here
https://github.com/ArduPilot/MissionPlanner/actions

jverhuls, 2021-01-19:
Nice Job! Is there a way how I can help testing?

meee1, 2021-01-19:
that depends on what you have access too.
the android version is in the google play store
ios and osx are only via github actions atm

jverhuls, 2021-01-19:
Ok thanks. I can test the OSX, iOS version. Will check if i can get them installed and working.

meee1, 2021-01-19:
[image]

as a side note, you can compile it under vs for mac as well

meee1, 2021-01-19:
you could test on a real iphone, that i have not tested.   iphone simulator it seems to work

jverhuls, 2021-01-19:
I can do that. Will build it first on mac (i'm installing the tooling).