ArduPilot/MissionPlanner #2293: Log Browser FOLL.Alt scaling error

twistedwings, 2020-01-09:
[image]

Values in the data grid are between -196 and 473 (cm)
Scaling should be in meters.

rmackay9, 2020-01-09:
What's odd is that it appears fine for me with MP 1.3.70 build 1.3.7312.21688.
[image]

Here's a link to the log: https://www.dropbox.com/s/9scyc81jyxf5e2l/twisted-wings-follow-log.bin?dl=0

meee1, 2020-01-09:
it appears correct for me as well.

did you load the bin or log file?

twistedwings, 2020-01-10:
I have loaded the bin file.
When loading the log file it looks like this:

[image]

meee1, 2020-01-10:
yes, it is highly advisable to always use the bin file.
there are a few quirks when using a log file due to double scaling.

rmackay9, 2020-01-10:
OK, I've reproduced this. I took the .bin file and used MP to convert it to a log file, then graphed FOLL.Alt and it looks like this:
[image]

I suspect the issue is that the FMT message for the FOLL message doesn't appear in the logs until lin 42941.

meee1, 2020-01-10:
check the MULT as well.

rmackay9, 2020-01-10:
I tried moving the FMT and FMTU messages up but it didn't help.  I don't see any different in the MULT messages..

meee1, 2020-01-10:
ok, found the issue.
please try the latest beta MP. (in about 15 mins)

twistedwings, 2020-01-10:
no change...

[image]

[image]

meee1, 2020-01-10:
are you in a country with a . vs , issue?

twistedwings, 2020-01-10:
I am in germany and have no idea what a . vs , issue is.

meee1, 2020-01-10:
https://en.wikipedia.org/wiki/Decimal_separator
germany, uses a period.
i will need to check this again to resolve the localization issue.

meee1, 2020-01-10:
could you send through the log file? not the bin. just want to see if the error is in converting to log, or reading the log

twistedwings, 2020-01-10:
this is the log:

[image]

rmackay9, 2020-01-10:
[image]
It looks ok to me now..

meee1, 2020-01-10:
i mean send the log via dropbox

twistedwings, 2020-01-10:
I'm sorry here it is:

[https://drive.google.com/open?id=1yGeebd7b-bl5KGkTUcZ-LuBn2mmYLpbn](url)

meee1, 2020-01-10:
ok new beta in another 15 mins
https://github.com/ArduPilot/MissionPlanner/commit/f5079406cee8f926b8964383af9ce48876d2b9ba

twistedwings, 2020-01-10:
issue is gone with latest beta...

[image]

[image]

meee1, 2020-01-10:
perfect.   the issue was around period vs comma's in the multiplier data

closed 2020-01-10 by meee1