ArduPilot/MissionPlanner #2296: MP 1.3.70  broken features and buttons under mono/linux

davidbuzz, 2020-01-13:
#### Issue details
tryijng to do a software update, beta software update, or start any SITL instance all appear to be broken for me on a newly downloaded 1.3.70 version and run under mono....  see three screenshots. 

#### Version
1.3.70 of MP downloaded as a zip from the firmware server , unzipped and then run with mono.

mono --version
Mono JIT compiler version 4.6.2 (Debian 4.6.2.7+dfsg-1ubuntu1)

#### Platform
[X  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
n/a'

#### Hardware type
n/a

#### Logs
_Please provide a link to any relevant logs that show the issue_
[image]
[image]
[image]

meee1, 2020-01-13:
the simulation tab will not work under linux (cygwin compiled exe's). id suggest manually starting it and connecting
the others i will check out

meee1, 2020-01-13:
i would also suggest upgrading mono to the 5.x series

meee1, 2020-01-13:
the update fail is around a tls issue.  do you have the current ca-cert installed?

davidbuzz, 2020-01-13:
hi and thanks for the suggestions..... 
regarding "i would also suggest upgrading mono to the 5.x series  " - i'd like to point out that i'm using a fully updated Ubuntu 18.04.3 LTS (just did software update this morning), so although i'm happy to try updating mono to a more experimental one as you suggest, the one i'm using *is* the latested one currently available in the ubuntu package repositories, so it's what most user/s will be using out-of-the-box.       maybe if MP needs mono 5.x to work, then it should insist on that at start-up time.   i will try using the 'mono' repo now to get v5 and report back.

davidbuzz, 2020-01-13:
followed instructions here to get latests stable mono...
https://www.mono-project.com/download/stable/#download-lin
ie:
sudo apt install gnupg ca-certificates
sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-keys 3FA7E0328081BFF6A14DA29AA6A19B38D3D831EF
echo "deb https://download.mono-project.com/repo/ubuntu stable-bionic main" | sudo tee /etc/apt/sources.list.d/mono-official-stable.list
sudo apt update
.... 
>165 packages can be upgraded. Run 'apt list --upgradable' to see them.
sudo apt upgrade  
... gets a bunch of libmono and mono package updates.

afterwards my mono version is now....
mono --version
Mono JIT compiler version 6.6.0.161 (tarball Tue Dec 10 10:36:32 UTC 2019)

after that i re-ran MP, and now the 'updates' and 'beta updates' both work.    i recommend MP checking on-startup that mono is NOT the default ubuntu version, and perhaps giving a link to where to get a newer version of mono from.
also, my mono is now gone from 4 to 6, not 5 as i thought it would.

davidbuzz, 2020-01-13:
regarding your comment "the simulation tab will not work under linux (cygwin compiled exe's)." - i'd like to suggest that rather than have all your future linux user/s try to run SITL and fail, that maybe you could add a simple check to see if the .exe is being run under mono, and either give a more meaning error with a documentation URL that tells them how to start SITL on linux, or maybe you can figure out how to embed a slightly different behaviour for linux based on testing if the .exe is being run under mono?

meee1, 2020-01-13:
yeah, agree. i never added any mono check in there. but would be easy to add

davidbuzz, 2020-01-13:
with the updated mono, the 'beta updates' button now presents a different issue... it can be pressed, and the update dialog starts trying to update, but does not seem to progress off zero....
[image]

davidbuzz, 2020-01-13:
... addendum to the last message, i waited approx 7 minutes from when i pressed the 'beta update' button and that dialog first appeared, and it did *eventually* change, and move onto the 'check for updates' dialog box.....    which then failed shortly there after... with this message...  
[image]

davidbuzz, 2020-01-13:
further... i closed MP, restarted it, and tried the 'beta updates' button again... and again after waiting about 9minutes i got this this similar but different error....
[image]

meee1, 2020-01-13:
im looking into the hashing issue now

davidbuzz, 2020-01-13:
thanks, i'd also recommend making the 'getting updated parameter documentation' dialog box tell the user that something is actually happening behind-the-scenes, as a typicall user isn't going to wait X(7-10 in my case) minutes with what looks like a 'hung' message and no on-screen activity.

davidbuzz, 2020-01-13:
another issue....  loading from file or reading mission from vehicle causes command column to show UNKNOWN under mono
[image]

davidbuzz, 2020-01-13:
i figured out error message/s behind the 'getting updated paramater documentation' ...
  at System.Net.HttpWebRequest.RunWithTimeoutWorker[T] (System.Threading.Tasks.Task`1[TResult] workerTask, System.Int32 timeout, System.Action abort, System.Func`1[TResult] aborted, System.Threading.CancellationTokenSource cts) [0x000f8] in <dfe42095fd31410e95d8021f755c3557>:0 
  at System.Net.HttpWebRequest.GetResponse () [0x00016] in <dfe42095fd31410e95d8021f755c3557>:0 
  at MissionPlanner.Utilities.ParameterMetaDataParser.ReadDataFromAddress (System.String address, System.Int32 attempt) [0x00151] in <123aa4e625ab43e981f7e71b791c57ba>:0 
ERROR MissionPlanner.Utilities.ParameterMetaDataParser - The request to https://raw.oborne.me/ardupilot/ardupilot/APMrover2-stable/libraries/AP_RPM/AP_RPM.cpp failed.
System.Net.WebException: The operation has timed out.
  at System.Net.HttpWebRequest.RunWithTimeoutWorker[T] (System.Threading.Tasks.Task`1[TResult] workerTask, System.Int32 timeout, System.Action abort, System.Func`1[TResult] aborted, System.Threading.CancellationTokenSource cts) [0x000f8] in <dfe42095fd31410e95d8021f755c3557>:0 
  at System.Net.HttpWebRequest.GetResponse () [0x00016] in <dfe42095fd31410e95d8021f755c3557>:0 
  at MissionPlanner.Utilities.ParameterMetaDataParser.ReadDataFromAddress (System.String address, System.Int32 attempt) [0x00151] in <123aa4e625ab43e981f7e71b791c57ba>:0 
ERROR MissionPlanner.Utilities.ParameterMetaDataParser - The request to https://raw.oborne.me/ardupilot/ardupilot/ArduPlane-stable/libraries/APM_Control/AP_RollController.cpp failed.
System.Net.WebException: The operation has timed out.
  at System.Net.HttpWebRequest.RunWithTimeoutWorker[T] (System.Threading.Tasks.Task`1[TResult] workerTask, System.Int32 timeout, System.Action abort, System.Func`1[TResult] aborted, System.Threading.CancellationTokenSource cts) [0x000f8] in <dfe42095fd31410e95d8021f755c3557>:0 
  at System.Net.HttpWebRequest.GetResponse () [0x00016] in <dfe42095fd31410e95d8021f755c3557>:0 
  at MissionPlanner.Utilities.ParameterMetaDataParser.ReadDataFromAddress (System.String address, System.Int32 attempt) [0x00151] in <123aa4e625ab43e981f7e71b791c57ba>:0 
ReadDataFromAddress attempt 0 Queued https://raw.oborne.me/ardupilot/ardupilot/APMrover2-stable/libraries/AP_BattMonitor/AP_BattMonitor_Params.cpp
ERROR MissionPlanner.Utilities.ParameterMetaDataParser - The request to https://raw.oborne.me/ardupilot/ardupilot/APMrover2-stable/libraries/AP_RSSI/AP_RSSI.cpp failed.
System.Net.WebException: The operation has timed out.

meee1, 2020-01-13:
latest beta should resolve all these issues

closed 2021-07-29 by davidbuzz