ArduPilot/MissionPlanner #2303: Load WP File shows scary error box when file is open in another program

rmackay9, 2020-01-20:
If a waypoint file is opened in another program (like Windows Excel) at the moment that "Load WP File" is used to attempt to load the file into the Flight Planner screen, MP shows a big scary error box.
[image]

I don't actually understand why MP can't open the file as read-only and avoid the error but in any case, if that can't be resolved then it should at least print a human friendly "Failed to open file, locked by another program" or something similar.

meee1, 2020-01-20:
the thing i take away from this is some other program has a read/write lock on the file. so MP cant not open it even for reading.

i will make the error a little less verbose however

rmackay9, 2020-01-20:
@meee1, I don't think excel has taken a read/write lock on the file.  It's only taken a write lock.  I can open the file in NotePad++ so I don't see why MP can't open it.  I suspect MP is trying to open it with a write lock.

rmackay9, 2020-01-29:
this seems to be resolved now, thanks!

closed 2020-01-29 by rmackay9