ArduPilot/MissionPlanner #2307: Where to edit parameter descriptions

SchrodingersGat, 2020-01-21:
I have recently been working on adding some new CAN features to the Ardupilot codebase, and would like to update the parameter descriptions in MissionPlanner.

e.g. `CAN_D1_PROTOCOL` now has extra values 

Where should I provide this extra information? I have found an `.xml` file called *ParameterMetaDataBackup.xml* file - is there a "master" location for this information?

Please let me know where to include this and I'll raise a PR.

Cheers

meee1, 2020-01-21:
on a MP update a new version file is generated based from ardupilot master.
check your my documents > mission planner > ParameterMetaData.xml for your current version.
is the new change in master yet?

SchrodingersGat, 2020-01-21:
@meee1 sorry I think I wasn't clear - I wish to *make* this new change so that the parameter file contains these new values. I cannot however find (in the source) the right location to add new parameters.

meee1, 2020-01-21:
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_BoardConfig/canbus_driver.cpp#L37

just do a beta update on your MP, and it will automatically add it. i assume you want PiccoloCAN

SchrodingersGat, 2020-01-21:
Ah, I see! I did not realize it got auto-generated from those docstrings. That makes sense! Thanks for your help.

closed 2020-01-21 by SchrodingersGat