ArduPilot/MissionPlanner #2308: System.EntryPointNotFoundException during update of MissionPlanner to Beta

gnitzan, 2020-01-21:
#### Issue details
_Please describe the problem, or desired feature_
Error dialog shown when trying to update to latest beta from latest stable version.

An error has occured
System.EntryPointNotFoundException: Unable to find an entry point named 'sk_refcnt_safe_unref'
in DLL 'libSkiaSharp'.

After this the MissionPlanner refuses to start.

#### Version
_What version was the issue encountered with_
1.3.7.0
#### Platform
[ X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_

#### Logs
_Please provide a link to any relevant logs that show the issue_

meee1, 2020-01-21:
please open the MP install directory and delete the file "libSkiaSharp.dll" . make sure its the one with "lib" in its name

gnitzan, 2020-01-30:
thank you @meee1 problem fixed

closed 2020-01-30 by gnitzan