ArduPilot/MissionPlanner #2309: After connecting over UDPCI the disconnect button no longer works.

sher1ock, 2020-01-23:
#### Issue details
After connecting or attempting to connect over UDPCI the disconnect button no longer works. It will say 'disconnected' very briefly when you try to disconnect but it stays connected. Issue appeared after updating from the auto updater. 
#### Version
Mission planner 1.3.70 Build 1.3.7277.34800
#### Platform
[  ] All
[  ] AntennaTracker
[ X] Copter
[ X] Plane
[  ] Rover
[  ] Sub

#### Airframe type
Any
#### Hardware type
Emlid Edge, but doesn't appear to be HW specific. 
#### Logs
N/A

meee1, 2020-01-30:
could you please try with the latest beta?

sher1ock, 2020-01-31:
Just tried with the beta and appears to be working properly again.