ArduPilot/MissionPlanner #2311: Actions ChangeSpeed button not working

rmackay9, 2020-01-25:
From real life and SITL testing it seems that MP's Action tab's ChangeSpeed button is not sending the correct message to cause the vehicle to change it's speed while in autonomous modes.

I've confirmed using mavproxy that AP is correctly accepting COMMAND_LONG messages with the command field set to MAV_CMD_DO_CHANGE_SPEED (178).

[Copter's implementation can be seen here](https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/GCS_Mavlink.cpp#L687).  It looks like the target speed should be placed in param2 and should be in m/s.

meee1, 2020-01-25:
This would have worked at some time. But newer firmware versions must have resulted in it ceasing to function

rmackay9, 2020-01-27:
Another user reported this issue during the Copter-4.0.1 release. https://discuss.ardupilot.org/t/copter-4-0-1-released/51505/4

rmackay9, 2020-02-06:
Re-tested with latest beta MP and it works, thanks!

By the way, sharing the number with the "Set Home Alt" is a bit naughty because it rounds off to the nearest integer.  I actually wonder why anyone needs to "Set Home Alt"... anyway, closing!

closed 2020-02-06 by rmackay9

nkikin, 2021-05-04:
Hi, I know this session is closed, but I just want to know which firmware version is required to get Actions changespeed button working?

rmackay9, 2021-05-04:
@nkikin, yes, this should work with Copter-4.0.x (stable) or Copter-4.1.0-beta1 (the current beta).  If it doesn't work with 4.1.0 could you report in the [Copter-4.1.0 beta testing category](https://discuss.ardupilot.org/c/arducopter/6)?  I'm quite actively monitoring reports.

nkikin, 2021-05-05:
@rmackay9 Thanks, it works with Copter-4.1.0.