ArduPilot/MissionPlanner #2315: DataFlash Err label is right justified instead of left justified

rmackay9, 2020-01-29:
While analysing a user log I was thrown off because the "Err: CRASH_CHECK -1" message appeared right-justified instead of left-justified.  This made me think that the crash had happened a lot earlier in the log than it actually did.  Below is a screen shot and you can see in the bottom "Data Table" area that the Error 12 appears at line 45670 but in the graph part at the top it appears from the label that it happens just before 44000.

[image]

rmackay9, 2020-01-29:
But now strangely when I graph the exact same log again the label is in the correct place.. i'm confused.
[image]

meee1, 2020-01-29:
it could be a drawing order issue. i will look further into it.

rmackay9, 2020-02-06:
I'm on the latest beta and I haven't seen this again so closing, thanks!

closed 2020-02-06 by rmackay9