ArduPilot/MissionPlanner #2319: geotiff Add in Gmap.Net MP

ashish-yadav1, 2020-01-31:
#### Issue details
_Please describe the problem, or desired feature_

#### Version
_What version was the issue encountered with_

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_

#### Logs
_Please provide a link to any relevant logs that show the issue_

meee1, 2020-01-31:
please search on the ardupilot forum. github is used for software issue tracking, not general support like this.
https://discuss.ardupilot.org/t/custom-imagery-in-mission-planner/10280/13
and
https://discuss.ardupilot.org/t/custom-maps-in-mission-planner/11290

closed 2020-01-31 by meee1

ashish-yadav1, 2020-01-31:
i have seen that but it use geoserver for wms custom  
can i directly while  geotiff file on  mp buy adding some thing in code?

meee1, 2020-01-31:
please check the control-f > custom gdal button
this will load all image types in a directory to show on the map.
then pick GDAL Custom from the map drop down in flight planner

ashish-yadav1, 2020-01-31:
thanks you so much