ArduPilot/MissionPlanner #2320: Adding new flight mode

Bggopal, 2020-01-31:
Hi guys,
I am trying to a new flight mode in Copter-3.5 using the documentation provided in ("https://ardupilot.org/dev/docs/apmcopter-adding-a-new-flight-mode-35.html"), but even after updating every field (by creating a new flight mode similar to stabilize) i couldn't see it in APM planner, i using APM planner v2.0.23 in linux. the most searchers where relating to a file in mission planner "ParameterMetaData.xml" but for me here it is "apm.pdef.xml" . Even though i edited it, it does not work. Was i on the right way, or where did i do mistake, if you any one have good experience, please provide your help here. Thank you.

Git repo for the changes i did "https://github.com/Bggopal/ardupilot/tree/Copter-3.5"

[x] Pixhawk
[x] Copter
[image]

rmackay9, 2020-01-31:
I guess this applies to the [APM Planner](https://github.com/ArduPilot/apm_planner/issues) and not Mission Planner?  I think I'll close this but here are some other places to find help:

- [APM Planner's issues list](https://github.com/ArduPilot/apm_planner/issues)
- [APM Planner's support area](https://discuss.ardupilot.org/c/ground-control-software/apm-planner-2-0/72)
- [Developer chat on gitter](https://gitter.im/ArduPilot/ardupilot).  In particular muramura and David Buzz know about APM Planner

closed 2020-01-31 by rmackay9

Bggopal, 2020-02-01:
@rmackay9  thank you Mr.