ArduPilot/MissionPlanner #2333: Log viewer resets DataTable/Graph split when window is made wider

rmackay9, 2020-02-14:
Making the log viewer window winder unnecessarily affects the position of the split between the graphing area (at the top) and the data table area (at the bottom).

Here is a picture of the window before:
[image]

And here it is a moment later after I have grabbed the right side and made it winder by a few pixels
[image]

meee1, 2020-02-14:
https://github.com/ArduPilot/MissionPlanner/blob/master/Log/LogBrowse.cs#L3165
atm when you resize the window it does a 50% split

rmackay9, 2020-02-14:
@meee1, yup, just not quite sure why it does that.  From a user point of view, making it wider shouldn't affect the vertical split.  Ideally it should maintain the current balance between up and down as the window is made wider or taller.