ArduPilot/MissionPlanner #2338: MP evaporates when connecting to latest AP

rmackay9, 2020-02-20:
I've found that MP disappears soon after attempting to connect to Copter-4.0.3-dev (aka "latest").  It's fine when connecting to Copter-4.0.2 (stable).  I suspect this is related to the new USB IDs used in master and the windows drivers installed using drivers.msi found in this directory: https://firmware.ardupilot.org/Tools/MissionPlanner/

meee1, 2020-02-20:
I believe this is caused by a recent windows update. I've tried net framework 4.8 and it appears to resolve it

meee1, 2020-02-20:
https://dotnet.microsoft.com/download/dotnet-framework/thank-you/net48-web-installer

rmackay9, 2020-03-03:
I can't recreate this problem any more so closing

rmackay9, 2021-06-20:
I'm reopening this because a number of users have reported this during 4.1 testing.