ArduPilot/MissionPlanner #2345: Bad logging after update from Copter 3.6.5 to 4.0.3

wcfung, 2020-03-03:
#### Issue details
_My 500 quad was using a clone Pixhawk running Copter 3.6.5 without any problem in logging both DF and .bin files. After updating to Copter 4.0.3 few days ago, I got "Bad logging" when arming motors. 
I then unchecked the "Logging available" option in the ARMING_CHECK param, but then logging is impossible. Tried many times to revert back to Copter 3.6.5, but MP was unable to detect the board, and reverting failed.

_ My OTHER quad 250 size is running a non-Pixhawk board (Open Pilot Revolution) using ChibiOS. Board is also flashed with Copter 4.0.3. I can arm this quad with logging DISABLED, see the link to screenshot below. Not even that, I can arm even BEFORE GPS IS LOCKED, as seen in the selected options of the screen shot below.  Can anybody explain ?  Thank you.

#### Version
_What version was the issue encountered with_
  Copter 4.0.3,   MP 1.3.70

#### Platform
[  ] All
[  ] AntennaTracker
[* ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
   Quad 500
 
#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_
  Pixhawk (clone)

#### Logs  (For Quad500)
_Please provide a link to any relevant logs that show the issue_(Quad__500, Pixhawk)
https://drive.google.com/file/d/1vjms9RXov81I2CslOFO8OmDwJO6C9tnF/view?usp=sharing
_Link to screenshot of ARMING _CHECK _(QUAD__500)
https://drive.google.com/file/d/15JNI0DtUSjbPFc4T_RbSB9AKblhGTyqK/view?usp=sharing

For Quad 250
_Link to screen shot of ARMING_CHECK options (Quad__250 , Revo board)
https://drive.google.com/file/d/169gjrwngTxLmIzlnFTeQBfU-LU-NPxZQ/view?usp=sharing

rmackay9, 2020-03-12:
Thanks for the report.  This is much more likely to be a flight code issue so it would be better to report it into the Copter-4.0 discuss area. https://discuss.ardupilot.org/c/arducopter/copter-40/155

I'll close the issue here if that's ok.

closed 2020-03-12 by rmackay9

wcfung, 2020-03-12:
That's ok. Thank you.