ArduPilot/MissionPlanner #2346: display named value float in Quick status

tridge, 2020-03-05:
It would be really useful to be able to display NAMED_VALUE_FLOAT values in the MissionPlanner "Quick" display. The NAMED_VALUE_FLOAT is a single float value in mavlink with a string name. If a user could add a "Quick" field and choose the name they want to display then that would allow us to send NAMED_VALUE_FLOAT from on-board lua scripts to display custom values. It would be a massively useful way of extending the ability of lua scripts.
There is one current NAMED_VALUE_FLOAT we send in ArduPilot, the "AS2" value which is the 2nd airspeed sensor. So in SITL you can enable a 2nd airspeed sensor and start getting AS2 NAMED_VALUE_FLOAT for testing.

EosBandi, 2020-03-21:
OK, should it be automatic, or something that we can define in the config ? I mean just wait for a mavlink message and add the named value to currentstate or we can predefine names in config and look for them when updating currentstate ?

meee1, 2020-03-22:
im thinking about using a dictionary in currentstate. but need to work though how complicated it will get

EosBandi, 2020-03-25:
https://github.com/ArduPilot/MissionPlanner/pull/2360 implements support fort NAMED_VALUE_FLOAT display.

IamPete1, 2020-07-16:
I have just tested this with, https://github.com/ArduPilot/ardupilot/pull/13753 Works great, thanks @EosBandi 

I think this issue can now be closed.

IamPete1, 2020-07-18:
Having finished messing with named floats I have discovered that it is impossible to remove it from the tuning graph. I had ticked it, now I just get 'customfield0' on the graph but no way to un-tick as its no in the list.

closed 2022-01-26 by meee1