ArduPilot/MissionPlanner #2348: Drone logbook for flight log management

Bornlove, 2020-03-08:
#### Feature details
_Please describe the problem, or desired feature_ the feature proposed will enable the flight log book to be pushed to that platform straight from the mission planner and will enable a pilot to keep all flight organized for easy follow-up of the flight hours for maintenance as well and so many benefits.

#### Version
_What version was the issue encountered with_ no any version has integrated this so far.

#### Platform
[√] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_ all airframe

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_ all hardware

#### Logs
_Please provide a link to any relevant logs that show the issue_ it is a request for new feature

meee1, 2020-03-08:
need more information

what information is required by the country/area you are?

Bornlove, 2020-03-10:
> need more information
> 
> what information is required by the country/area you are?

Flight log Management

Bornlove, 2020-03-10:
> need more information
> 
> what information is required by the country/area you are?

https://www.dronelogbook.com/hp/1/index.html
The feature focused from this link is for post flight log management. Will this a pilot will keep him/ herself on the track with the logfiles. If there is any means of doing so without interesting dronelogbook, please let me know so as to close the issue raised. Thanks

closed 2020-03-10 by Bornlove