ArduPilot/MissionPlanner #2353: Bad right-click menu colors

AndKe, 2020-03-11:
#### Issue details
Bad right-click menu colors

#### Version
Current "advanced" installation: build 1.3.7277.34800  on MONO  
(fresh download)

#### Platform
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
N/A

#### Hardware type
n/a

#### Logs
[image]

meee1, 2020-03-12:
What mono version?

AndKe, 2020-03-12:
Sorry for forgetting that information.
```
Mono JIT compiler version 5.18.0.240 (Debian 5.18.0.240+dfsg-3 Mon Jun 24 08:50:44 UTC 2019)
Copyright (C) 2002-2014 Novell, Inc, Xamarin Inc and Contributors. www.mono-project.com
	TLS:           __thread
	SIGSEGV:       altstack
	Notifications: epoll
	Architecture:  amd64
	Disabled:      none
	Misc:          softdebug 
	Interpreter:   yes
	LLVM:          supported, not enabled.
	Suspend:       preemptive
	GC:            sgen (concurrent by default)
```

And thanks for finally trying to go multiplatform. (at least I see a lot more that works now, than before.)

meee1, 2020-03-16:
can you try mono in the 6.x series?
[image]

`Mono JIT compiler version 6.8.0.105 (tarball Tue Feb  4 21:20:20 UTC 2020)
Copyright (C) 2002-2014 Novell, Inc, Xamarin Inc and Contributors. www.mono-project.com
        TLS:           __thread
        SIGSEGV:       altstack
        Notifications: epoll
        Architecture:  amd64
        Disabled:      none
        Misc:          softdebug
        Interpreter:   yes
        LLVM:          yes(610)
        Suspend:       hybrid
        GC:            sgen (concurrent by default)`

AndKe, 2020-03-16:
Yes, I did discover that current stable 6.8.0 does look good.
There are a number of ways to crash it (AP) at will.
Especially when playing with logs, opening/converting - would you fix If I report them?

closed 2020-03-16 by AndKe

meee1, 2020-03-16:
please do report them, but also make sure you running the beta version
https://ci.appveyor.com/api/buildjobs/v44d3jo4vf76jnof/artifacts/MissionPlannerBeta.zip