ArduPilot/MissionPlanner #2358: display AIS vessels

IamPete1, 2020-03-21:
It would be great to display AIS vessels in a similar way to ADSB aircraft. I have done a ArduPilot and Mavlink PR to add this functionality. I did make a attempt to build mission planner locally but did not have any success.

https://github.com/ArduPilot/mavlink/pull/131
https://github.com/ArduPilot/ardupilot/pull/13839

meee1, 2020-03-21:
@IamPete1 What items should i show/be useful?
   <message id="301" name="AIS_VESSEL">
      <wip/>
      
      <description>The location and information of an AIS vessel</description>
      <field type="uint32_t" name="MMSI">Mobile Marine Service Identifier, 9 decimal digits</field>
      <field type="int32_t" name="lat" units="degE7">Latitude</field>
      <field type="int32_t" name="lon" units="degE7">Longitude</field>
      <field type="uint16_t" name="COG" units="cdeg">Course over ground</field>
      <field type="uint16_t" name="heading" units="cdeg">True heading</field>
      <field type="uint16_t" name="velocity" units="cm/s">Speed over ground</field>
      <field type="int8_t" name="turn_rate" units="cdeg/s">Turn rate</field>
      <field type="uint8_t" name="navigational_status" enum="AIS_NAV_STATUS">Navigational status</field>
      <field type="uint8_t" name="type" enum="AIS_TYPE">Type of vessels</field>
      <field type="uint16_t" name="dimension_bow" units="m">Distance from lat/lon location to bow</field>
      <field type="uint16_t" name="dimension_stern" units="m">Distance from lat/lon location to stern</field>
      <field type="uint8_t" name="dimension_port" units="m">Distance from lat/lon location to port side</field>
      <field type="uint8_t" name="dimension_starboard" units="m">Distance from lat/lon location to starboard side</field>
      <field type="char[7]" name="callsign">The vessel callsign</field>
      <field type="char[20]" name="name">The vessel name</field>
      <field type="uint16_t" name="tslc" units="s">Time since last communication in seconds</field>
      <field type="uint16_t" name="flags" enum="AIS_FLAGS" display="bitmask">Bitmask to indicate various statuses including valid data fields</field>
    </message>

IamPete1, 2020-03-21:
I guess Mobile Marine Service Identifier, speed and turn rate.  I guess location and heading are sort of inherent in the display.

Some of the other info might be interesting, Type of vessel for example but due to the way AIS works that info is only sent occasionally so it is likely to be missing. The location speed, heading and turn rate are sent much more frequently.

Thanks for talking a look at this so soon!

IamPete1, 2021-11-17:
Should probably have close this ages ago....

https://youtu.be/pGluWMlIXho

closed 2021-11-17 by IamPete1