ArduPilot/MissionPlanner #2362: HUD consumes 100% CPU

expntly, 2020-04-05:
#### Issue details
When MP starts by default and with the HUD visible, after connecting to a vehicle the Windows Task Manager is telling me MP is consuming 100% CPU.

When I hide the HUD (double click on it and minimize), CPU immediately goes back down to near 0%.

When at 100%, the UI in MP feels slow and sluggish, buttons take time to click, it lags to go from one tab to another, etc.

I took the habit of minimizing the HUD as soon as MP starts and only show it when I fly, but it would be great if the HUD wouldn't consume that much CPU.

#### Version
All versions, the one I'm using now is 1.3.70 build 1.3.7277.34800

#### Platform
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

meee1, 2020-04-05:
please update your graphics card driver.
i have seen issues with bad drivers on intel cards before.
the hug uses opengl, and requires hardware acceleration

PRIMETSS, 2020-05-04:
Noticing this to on new build of 1.3.70 1.3.7428.24840 on laptop that was running server versions previous.
When going to Data Page, HUD displays, but other windows just go blank for 20-30 seconds. Seems sluggish.
Intel Graphics, latest driver (not changed recently, but have installed Windows Updates)
Its not a high performance laptop Intel i5 but not noticed this before in MP

Work around for now is to select Use Old GDI HUD in settings
[image]

meee1, 2020-05-04:
i would suggest installing the intel driver, not the windows update driver. that could fix a few issues

PRIMETSS, 2020-05-04:
Yes might be right

Found older version I was using before and ran it again, and seems CPU it was also High on that, just didnt notice any (obvious) menu lockups. Yes might be just performance issue. Desktop PC handles HUD OK. Trying another more up to date laptop to see if its just this one laptop (now)
[image]

meee1, 2020-05-04:
turn on the console window
help > show console window
and after restarting you will see
HUD 1 hz drawtime 3 gl True

and let me know what you see for drawtime and gl

PRIMETSS, 2020-05-04:
Attached HUD visible but menu still not drawn yet (10-20seconds after connecting, and Params updated)
[image]

meee1, 2020-05-04:
my pc takes 1-3 ms to draw the hud.
yours is taking 170ms to draw the hud.  definitively a graphics card issue.
try some other drivers to improve it.

PRIMETSS, 2020-05-04:
OK thanks, yes On my DEV machine which doesnt have this issue 
[image]

So must just be noticible on the laptop Ive been using for ages, either just hit a performance threshold now, or might be some as you say dodge driver or maybe windows updates.
I can cope tihe old GDI on that for testing. The Field laptop is a ToughBook, so should perform OK on that.

Thanks! For your time! Appreciated!!!!