ArduPilot/MissionPlanner #2363: Can't enter fractional speeds in Actions/Change Speed

cjmssmd, 2020-04-06:
#### Issue details
I'd like to be able to tell my mower to change from 1 to 1.5 m/s using Mission Planner.  But I can't since that entry box seems to limit to only integers., forcing me to decide between 1 and 2 m/sec.

#### Version
1.3.7314_23103

Steve-Scope, 2022-06-07:
I have a similar issue when planning a mission for our boat. Would be great to be able to set a speed between 1 and 2m/sec.
1m/sec is probably ideal for bathymetric surveying, but just a bit too slow for big jobs; and 2m/sec starts to chew the battery too much. 
Would this be something anyone was interested to tackle? Allowing the speed settings to have one decimal place??

meee1, 2022-06-07:
which box are you refering too?

Steve-Scope, 2022-06-07:
When planning a mission using 'AutoWP', Simple Grid, One of the settings you need to specify is the mission speed. It's in the 'Simple Options' box. See attached
[image]
.