ArduPilot/MissionPlanner #2365: GeoFenceDist no longer available on Quick Tab "Display This" list of Items

Warren-Eather, 2020-04-09:
#### Issue details
GeoFenceDist no longer available on Quick Tab after upgrade.
We have been using it for Fence Proximity Warnings and in Scripted Soft Fence Avoidance for the OBC

#### Version
MP 1.3.70 Build 1.3.7277.34800
[image]

#### Platform
[ x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

SSUAV, 2020-05-28:
Hi,
Any update on this?
Regards 
Warren

meee1, 2020-05-28:
i will look at adding this back, it was removed because with the new geofence support it got alot more difficult. but dont see any roadblocks bringing it back, other than time

meee1, 2020-05-28:
please try the latest beta. should be back and support multiple inclusion/exclusion zones

Warren-Eather, 2020-05-28:
Thanks for that Michael.
It appears that the Geofence Return point is also being included in the distance calculation. [cid:E25854C1-91EE-421F-8F3B-3E6D0DC9AA84-L0-001]
Warren

meee1, 2020-05-28:
ok new beta again, ignores return point

meee1, 2020-05-28:
ok ive checked it and im happy
[image]

Warren-Eather, 2020-05-29:
Hi Michael,

It appears that the GeoFence Return is still being included in the fence distance calculations

Regards

Warren

[image]<https://user-images.githubusercontent.com/511568/83157851-a4002400-a136-11ea-9f76-341e8026a374.png>

meee1, 2020-05-29:
do you have the latest beta? all my testing says its fixed

Warren-Eather, 2020-05-29:
1.3.71.1 build 1.3.7454.12815

meee1, 2020-05-29:
i think i need to see your fence. as its good from what i can see
also what firmware and version.

Warren-Eather, 2020-05-29:
I'm currently just testing on the MP simulator.

meee1, 2020-05-29:
what firmware? copter/rover/plane.

Warren-Eather, 2020-05-29:
Oh soz, Plane

SSUAV, 2020-06-04:
Thanks Michael 
Looking good!
Regards 
Warren
SSUAV

meee1, 2020-06-04:
yeah, fixed the other day, forgot to mention it. was the fact that plane still uses the old legacy fence protocol. the others use the new mission item one. but both should be good now

closed 2020-06-04 by meee1