ArduPilot/MissionPlanner #2368: 2 OverShoot values, but only 1 LeadIn value ??

jagrandle, 2020-04-12:
#### Issue details
In MP there are two OverShoot values, one for each end of the survey grid. This is great. However, there is only one LeadIn value so this applies to both ends of the grid.

In windy conditions with a fixed wing it may be necessary to go a long distance up wind to allow sufficient distance to turn around before re-entering the survey area. Conversely down wind there may not need to be very much distance at all to turn around. It wastes a lot of time having to travel the same distance down wind due to the single LeadIn value.

To get the best turn performance this needs both OverShoot and LeadIn.

Please consider splitting LeadIn in to two values, one for each end of the survey grid, just like OverShoot.

Thank you.

#### Version
1.3.70 currently but it has the same for a long, long time.

#### Platform
[X] All
[  ] AntennaTracker
[  ] Copter
[X ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
Mainly fixed wing but should not be aircraft specific

#### Hardware type
n/a

#### Logs
n/a

meee1, 2020-04-18:
the original reason for this was to do with the ability to create a predictable approach on entry.
ie
overfly on the way out, giving more time to get back online for the leadin.

jagrandle, 2020-04-18:
Thanks for the reply.
Given that the code must already have a way to differentiate the two ends of the flight plan in order to allow the two different values of OverShoot, would it be possible to do the same for LeadIn ? 
In stronger winds, with lots of rows, lots of time is wasted going too far down wind with the current options.
Thank you.