ArduPilot/MissionPlanner #2369: ASL Altitude inaccurate in flight plan

jonasbouchraiet, 2020-04-14:
When I check the altitude of a few mountain peaks in my sorroundings the number is about 300m too low compared to the Actual Altitude above sea level.

e.g. Hover over the peak of the Mount Everest with your curser: It shows about 8500 Meters. It's actually 8848m ASL.

In google earth Pro it seems to be right with an Accuracy of 1m.

Am I missing something?

meee1, 2020-04-15:
what lat/lng are you comparing?

jonasbouchraiet, 2020-04-15:
lat: 46,6120684
long: 9,6712518

Mission Planner: 2675m ASL
Google Earth pro: 3146m ASL
Difference: 471m

or:

lat: 46,3823650
long: 9,9081284

Mission Planner: 3610m ASL
Google Earth pro: 4050m ASL
Difference: 440m

meee1, 2020-04-15:
looks like srtm data is not very accurate in this area.
[image]
over all the image looks correct, however i cant explain the elevation error at this point in time.

jonasbouchraiet, 2020-04-15:
So... Does Google Earth Pro use another elevation measuring method or is it just different srtm data?

meee1, 2020-04-18:
https://www.ncbi.nlm.nih.gov/pmc/articles/PMC5296860/
 However, vertical accuracy of the data decreases with increase in slope and elevation due to presence of large outliers and voids. Therefore, studies using SRTM data “as is”, especially in regions like the Himalaya, are not statistically meaningful. 

google must source there data from elsewhere for these areas.

meee1, 2020-04-18:
if you can find anotehr source, Mission planner does support custom DTM's/DEM's if this is required