ArduPilot/MissionPlanner #2370: Connect doesn't start until "Timeout in 7s"

expntly, 2020-04-16:
#### Issue details

When connecting to the drone, it is not until 23s after and the pop up window showing "Timeout in 7s" that we finally start receiving telemetry and downloading the params.  It is systematically the same behavior, nothing happens before it shows "7s".

#### Version
1.3.68.1

#### Platform
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Hardware type
PX4

meee1, 2020-04-18:
what type of connection. you have not provided alot of details

expntly, 2020-04-18:
RF connection to the drone.  So it's on COM8 or something, and 115200 baudrate.  RFD900+.

Have you seen this before?

What can I do to debug this issue?  It looks like MP is just sitting there waiting until the 23rd second to get started.

expntly, 2020-04-27:
Baudrate is actually 57600, in case this helps.  Is there a way to see what MP is doing for 23 seconds?