ArduPilot/MissionPlanner #2372: three errors at the  first time I open the project

joelianruntu, 2020-04-20:
#### Issue details
Hi, bro. I'm a green hand so I have some questions to solve. I had download the lastest project with git fully, and I had also installed VS2017 and selected those necessary. But the IDE found that there are still 3 errors :

[1] ERROR 	NETSDK1045	Your Net SDK does not support setting**. Net core 3.1** as a target. Please target. Net core 2.2 or earlier, or use a. Net SDK version that supports**. Net core 3.1.**	px4uploader	C:\Program Files (x86)\dotnet\sdk\2.2.110\Sdks\Microsoft.NET.Sdk\targets\Microsoft.NET.TargetFrameworkInference.targets	137	
[2] ERROR	NETSDK1045	Your Net SDK does not support setting**. Net core 3.1** as a target. Please target. Net core 2.2 or earlier, or use a. Net SDK version that supports**. Net core 3.1.**		Ntrip	C:\Program Files (x86)\dotnet\sdk\2.2.110\Sdks\Microsoft.NET.Sdk\targets\Microsoft.NET.TargetFrameworkInference.targets	137	
[3] ERROR	NETSDK1045	Your Net SDK does not support setting**. Net standard 2.1** as the target. Please set. Net standard 2.0 or lower as the target, or use a. Net SDK version that supports**. Net standard 2.1.**	wasm	C:\Program Files (x86)\dotnet\sdk\2.2.110\Sdks\Microsoft.NET.Sdk\targets\Microsoft.NET.TargetFrameworkInference.targets	150	

I know from the Internet that **.net core 3.1** doesnot support VS2017. I donot know what should I do to make "0 error". If you know it, please tell me. Thanks very much!
Know little about Markdown, sorry for the format.

#### Version
Mission Planner Ground Control Station (c# .net)   master

#### Platform
[ 1 ] VS2017 Community

meee1, 2020-04-20:
Upgrade to vs2019

joelianruntu, 2020-04-20:
thanks，but when I try it on vs 2019（had tried）, there is still an error:
"filetest" CANNOT start with  a  lowercase letter ，should I modify it？

meee1, 2020-04-20:
build just mission planner. not the entire solution

meee1, 2020-04-20:
or rename it yes

joelianruntu, 2020-04-20:
ok ,I will try to build just MP today(not now, vs2019 is not in my PC but my friend's). so if I try to rename it, I only need to  rename the file "filetest.razor"  and it will work ,right?

lelabs, 2020-04-23:
Hey guys, renaming to Filetest.razor fixes this issue, but even with VS2019 it's not possible to compile mission planner only because of this error:
Your project does not reference ".NETFramework,Version=v4.6.1" framework. Add a reference to ".NETFramework,Version=v4.6.1" in the "TargetFrameworks" property of your project file and then re-run NuGet restore.	MissionPlanner	

Tried the googled hints without luck
- delete bin and obj folders
- install nuget restore
- change to .NETFramework v4.6.2 
- delete and fresh clone of the git repo 	

still the same error, 
@joelianruntu are you able to compile meanwhile?