ArduPilot/MissionPlanner #2375: Tilt trim not working

nagordon, 2020-04-25:
#### Issue details

using Optional Hardware -> Antenna Tracker Maestro, the tilt does not work with the antenna tracker. The trim does not seem to work either. I can adjust the the tilt by changing the "Center PWM" range, but that is all. I can control the antenna tracker fine with Pololu Maestro Control center.

I am using sitl to test and tried tlogs

#### Version
1.3.70 build 1.3.7418.41618

[image]
[image]

meee1, 2020-04-28:
the settings in the maestro app are not used when using the interface, the idea is you need to specify the range and center in the ui.
eg
pan
180 - seems good
range seems low, i would expect   1920-1040 = 880
center 1500 is good

then the same for tilt
angle is good
pwm range seems way to small   1632-1376 = 256
center should be 1504

the trim adjust the angle in degrees that the tracker is out.

meee1, 2020-04-28:
check here http://vps.oborne.me/gcs/Tracker.html

meee1, 2020-04-28:
one extra note, if you home location and vehcile are within 5m, then the tracker will not move.

nagordon, 2020-04-29:
awesome, i will check it out in the next few days

nagordon, 2020-04-30:
Well, I tried setting up with the settings as suggested. The endpoints are set so as not to crash, so they are basically fixed. Now, both trims work, but the antenna is no where close to the plane. when I adjust the center pwm and range to get the right "gain" so it actually tracks the aircraft, I am back where I started. It also seems to get stuck when it is at its pan endpoints. I suppose the best solution is to just use a controller that can actually sense the location. 

[image]
[image]

I also have identical results when I use an arduino, so clearly I do not have the settings correct

meee1, 2020-04-30:
Leave the centre alone. Only change trim

nagordon, 2020-04-30:
hmm, when i do that, it overshoots when tracking.

meee1, 2020-04-30:
define what you mean by overshot?

meee1, 2020-04-30:
pwm range = angle

ie min pwm = 0 deg
max pwm = angle

nagordon, 2020-04-30:
the change in pwm signal does not equal the correct number of degrees in pan or tilt

meee1, 2020-04-30:
so given you pwm range, what is the actual movement in degrees?

nagordon, 2020-04-30:
for pan it is short of the full + and - 180 mark. for tilt, although the range allows it, when i "fly" overhead it does not go vertical.

meee1, 2020-04-30:
ok pan angle = 350? not 180

nagordon, 2020-04-30:
possible, i need to check it. So I should use the angle to "tune" the endpoints? anyway, i am willing to accept that this is likely not a bug and just a user error and i can close if you concur.

meee1, 2020-04-30:
sounds good. post on the ardupilot forum if you have more questions, and i will answer there.
also allows others facing the same issue to search and find answers.
in a nutshell
your angle = the physical movement from pwm min to pwm max. in degrees

closed 2020-04-30 by meee1