ArduPilot/MissionPlanner #2376: Feature request:  Calibrating Pixracer compass over UDP

mebj, 2020-04-27:
#### Issue details
When calibrating the Pixracer compasses via UDP (i e using the Pixracer WiFi functionality) the only available option is the "live calibration" which the documents say is less accurate. On the other hand, when connecting via USB the "onboard calibration" is the only possible alternative.

I know that the onboard calibration using stick gestures is an option but it is a lot easier to follow the progress of onboard calibration if this worked also over UDP. 

I have seen a similar question about this but not found an answer anywhere.

#### Version
Mission Planner 1.3.7277.34800

#### Platform
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
F450 type quad frame
#### Hardware type
Pixracer R14

#### Logs
_Please provide a link to any relevant logs that show the issue_

meee1, 2020-04-28:
Please update the firmware on your esp8266

mebj, 2020-04-28:
Thank you for taking the time to comment. 

I thought that all my ESP boards were up to date (have 3 of them and have done some updating) and could not guess that this could be the root of the problem since all worked without any specific errors.

I have updated the firmware tonight and the onboard calibration shows up also when using the UDP connection. I will test the functionality tomorrow but it is presumably correct .

mebj, 2020-04-30:
Tested yesterday, onboard calibration working after update of ESP firmware.

closed 2020-04-30 by mebj