ArduPilot/MissionPlanner #2377: Hide RC failsafe in HUD when it in AUTO or GUIDED

jaguraphg, 2020-04-28:
#### Issue details
Please do not show RC failsafe message on HUD when my flight is on AUTO or GUIDED mode while the  FS_OPTIONS parameters say to one of the followiing:

bit 0 set: Continue if in auto mode on Radio Failsafe
bit 2 set: Continue if in guided mode Radio Failsafe

Reason: I do the long range flights, I do need the radio to take off and land manually close to home. But when I am in the air I execute the mission and my quad is flying away to the mission and I do not want to see the FAILSAFE message across entire flight. Thanks.

#### Version
1.3.70

#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
QUAD

#### Hardware type
Holybro Kakute F7

#### Logs
_Please provide a link to any relevant logs that show the issue_

jagrandle, 2020-04-28:
Maybe this should be an option rather than just removing it ??

Some users might find it useful to know the range performance of their radio from when the Failsafe is finally recorded.

It might also be useful know when you are back within signal range, by the removal of the Failsafe warning. If the radio link is not regained when expected it may give you more time to undertake remedial action before it arrives overhead, ready to land.

SSUAV, 2020-04-28:
Agreed!!!

jaguraphg, 2020-04-28:
I agree, this makes sense to make it configurable