ArduPilot/MissionPlanner #2379: Cannot set RCx_Option

ThisIsTheOnlyUsernameAvailable, 2020-05-03:
#### Issue details
Cannot set RC7_Option.  Also happens on RC8 and RC9, but others seem ok.
MissionPlanner reports "Set rc7_option Failed"

It isn't the specific option I'm choosing (tried lost model alarm, emergency motor stop, and others).

QGroundcontrol successfully sets the option.

#### Version
MatekF405 00240034 31374705 39373630
ChibiOS: d4fce84e
ArduCopter V4.0.3 (ffd08628)
Mission Planner 1.3.70 build 1.3.7426.10530 (ffd08628)

#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
X4

#### Hardware type
Matek F405-CTR

#### Logs
INFO MissionPlanner.MAVLinkInterface - set giveComport Base Thread current False new True
INFO MissionPlanner.MAVLinkInterface - setParam 'RC7_OPTION' = '30' sysid 1 compid 1
bps 1711 loss 14 left 111 mem 127.3994140625 mav2 True sign False mav1 0 mav2 53 signed 0      
INFO MissionPlanner.MAVLinkInterface - setParam Retry 3
bps 1711 loss 14 left 78 mem 127.3994140625 mav2 True sign False mav1 0 mav2 53 signed 0      
INFO MissionPlanner.MAVLinkInterface - setParam Retry 2
bps 1862 loss 14 left 144 mem 127.3994140625 mav2 True sign False mav1 0 mav2 58 signed 0      
INFO MissionPlanner.MAVLinkInterface - setParam Retry 1
INFO MissionPlanner.MAVLinkInterface - set giveComport Thread Pool Worker current True new False

meee1, 2020-05-03:
could you provide a tlog with the attempt?

ThisIsTheOnlyUsernameAvailable, 2020-05-03:
[tlog.zip](https://github.com/ArduPilot/MissionPlanner/files/4569534/tlog.zip)

Thanks, Michael.  Please see attached.

meee1, 2020-05-03:
what screen where you using to set this?

ThisIsTheOnlyUsernameAvailable, 2020-05-03:
The full parameter list

meee1, 2020-05-03:
ok the issue is there is a value being appended to the send for some reason.
i will look further
eg

RC7_OPTION
--
RC7_OPTION
RC7_OPTIONe
RC7_OPTIONZ
RC7_OPTIONZ
RC7_OPTIONZ
RC7_OPTIONe
RC7_OPTIONZ

ThisIsTheOnlyUsernameAvailable, 2020-05-03:
Thanks very much!

meee1, 2020-05-03:
tracing it down
so far looks like a mono bug... and old one
https://stackoverflow.com/questions/16966346/is-it-unsafe-to-use-tochararray-on-a-string-under-mono

ThisIsTheOnlyUsernameAvailable, 2020-05-03:
Sorry to have wasted your time.  I'll investigate resolving the Mono bug.

meee1, 2020-05-03:
im not 100% on that yet. so still looking.

meee1, 2020-05-03:
i think it might be a MP bug. its just odd how the first attempt works. and the second fails

meee1, 2020-05-03:
either way i would like to resolve it so MP works under mono

ThisIsTheOnlyUsernameAvailable, 2020-05-03:
Are you seeing the same behavior as I am?  Is there anything I can do to help further debug?

meee1, 2020-05-03:
yes, been able to replicate. 
its definitely something to do with mono. but like i said, my plan is to make it work.

meee1, 2020-05-03:
ok, please try the latest Beta MP.
this should be resolved.

ThisIsTheOnlyUsernameAvailable, 2020-05-03:
That fixed it.  Thanks so much for your help!

closed 2020-05-03 by ThisIsTheOnlyUsernameAvailable