ArduPilot/MissionPlanner #2384: RC_Options available on some computers but not others ??

jagrandle, 2020-05-11:
#### Issue details
I have assembled quite a few aircraft and multis with black cubes.

Now I have my first Orange one, though I can't see how colour gives me the problem I have.

Essentially I can't set channels 7,8,9, or 10 to do anything in either the Extended Tuning page or User Params page as all the options are greyed out. This is running copter v4.0.3 on orange. A colleague has same version running on black with options available.

After not receiving any workable replies on Facebook I tried some more.

I downloaded 1.3.70 to my laptop. It was a fresh download but it is the same build number that I had been using on my desktop till now. I went to Extended Tuning and there were now heaps of options listed for each RC Channel. I went to the Full Parameter Tree and in the Range column there is a description of all the mode numbers and what they correspond to. I set it to #17 and wrote the parameters. If I go to User Params then they are all greyed out and still nothing is selectable.

I then took the same orange cube back to my desktop. In the Full Parameter tree the RC_Option is still set to 17, but there is no description in the Range column. When I go to extended Tuning it does NOT show the option I have selected and they are still all greyed out. How can it be so different from one computer to the other with the same software ??

Screen shots and parameter file below (before I set RC7 to Type 17)
Thank you

[image]
[image]

[Orange Cube 2020-05-07.zip](https://github.com/ArduPilot/MissionPlanner/files/4608659/Orange.Cube.2020-05-07.zip)

#### Version
1.3.70 build 1.3.7277.34800

#### Platform
[  ] All
[  ] AntennaTracker
[ X ] Copter   v4.0.3
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
Quad

#### Hardware type
Orange Cube

#### Logs

meee1, 2020-05-11:
Have you tried mission planner beta?

jagrandle, 2020-05-12:
Hi. I wasn't sure how to reply so I tried replying to yourself from the e-mail at  notifications@github.com  but that doesn't seem to have worked !!

No, I have not tried Beta.

I don't really want to use Beta versions as I can't really afford for there to be other problems (like when the Polygon tool suddenly disappeared from the right click menu !!)  but I could try it if I can install it as an additional copy, and it might help you fix the problem.

How do I install Beta ?

In MP I went to the Help page then selected Check for Beta Updates. Unfortunately I got an error. Screen shot attached.

I also tried looking at the place where I usually download the MP versions so I can install them in the location I went without disturbing older versions, but there was nothing that indicated what was a Beta version.

In the Archive folder there are versions 1.3.8 and 1.3.9 but they are dated 2016 so that is confusing ??

Thanks,
Jeremy

[image]

meee1, 2020-05-16:
please try run as admin to update given that error