ArduPilot/MissionPlanner #2387: LogBrowse: decimal scalar fails on separator other than „.“

vierfuffzig, 2020-05-13:
Issue details
setting a decimal scalar in LogBrowse requires system (windows) separator set to „.“ instead of accepting system regional default

Version
1.3.7426.10530

Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x] Plane
[  ] Rover
[  ] Sub

meee1, 2020-05-16:
can you clarify this. i  kinda of understand, but want to make sure im not asuming things

whats your system set to? comma's or periods?
what are you entering?
anything else thats relevant

vierfuffzig, 2020-05-17:
@meee1 thanks for getting back on this!
my system's regional setting is comma per default. with this system setting, LogBrowse decimal scalar fails on commata and ignores periods, so that entering `*i,d` has no effect at all while `*i.d` is treated as `*id`.
only with the system set to period this works as i would expect (correct with periods, fail on commata).
...hope i managed to not add confusion ;)

vierfuffzig, 2020-06-17:
@meee1 fixed, thanks a lot!

closed 2020-06-17 by vierfuffzig