ArduPilot/MissionPlanner #2389: (BUG) Mission Planner will not set "new" home position for itself, but do so on FC, if trigger from Flight data screen.

pompecukor, 2020-05-16:
#### Issue details
When "Set Home Here" is used during a flight, in the flight data context menu (right click on map area, see picture below). Then the "new" home is sent and set on the flight controller, but mission planner itself is not updated.
You will see the "H" icon still showing (on the map) at the old location and also "distance to home" will report the distance of UAV to old location.
This information is misleading and there is no message to know that home has changed.
However the easy way to realize this is hit RTL and see the UAV flying to the "new" home location.
Setting home from Flight PLAN view, does work correctly, this problem only occurs if set from Flight DATA tab.

#### Version
Tested on Lastest beta version of MP: 1.3.7277.34800

#### Platform
Not relevant

#### Logs
No logs needed. Can reproduce easily in MP SITL...

Just select airplane aircraft in SITL, make a take off and few waypoints to get you air born
Arm, you see HOME set to arming position if it was not there otherwise. This is correct.
Now while in the air. Set new HOME position with the context menu in Flight data TAB.
You will see the the H icon on the map will not change, not will your distance to home. They will all show distance to the original HOME.
Now hit RTL, you will see the vehicle flying the the new position that you set, meaning that the FC knows about the new home, but MP does not, even tough MP was used to set it.

[image]

meee1, 2020-05-16:
ok so it depends how you use this
what you need to do is already be armed for this to stick.

if you do it while disarmed, it sets home, but as soon as you arm the drone will overwrite the home location with the armed location.

so you must do it after you have armed.

pompecukor, 2020-05-16:
> ok so it depends how you use this
> what you need to do is already be armed for this to stick.
> 
> if you do it while disarmed, it sets home, but as soon as you arm the drone will overwrite the home location with the armed location.
> 
> so you must do it after you have armed.

This happens while vehicle is flying. That is it is obviously armed.

When you arm (before flight) the FC will set home correctly. However when flying and using this context menu then what I described in details above will happen.

Apart from fixing, it would be a welcome change if setting new home from here in flight would need a confirmation, just as arming does. Just as saftly precaution.

pompecukor, 2020-05-17:
@meee1 
Would you like me to make a short video, to better demonstrate this to you?

jagrandle, 2020-05-18:
I agree. This has been an issue  for a long time. I am currently using 1.3.70  To my mind the Home position should be properly setable either via the Flight Data Page or the Flight Plan page. When it is changed in one it should update in the other.

Currently when you do it on the Flight Data page I thought nothing actually happened. pompecukor is suggesting that something does, but if it does then it is not shown.

This would also make flight planning easier as once you have found your location on the Flight Data page you can set your Home, then when you go to the the Mission Planning page either it would open with the map centred there, or at least you could zoom out till you could see the Home marker, then zoom back in on it,. And also vice versa.

Cheers Jag

pompecukor, 2020-05-18:
@meee1 

I made video: https://youtu.be/YuuHrbAKAAQ

pompecukor, 2020-06-05:
@meee1 
I just tested what you did. Excellent job Mike as usual. Thank you for the fix.

closed 2020-06-08 by meee1