ArduPilot/MissionPlanner #2396: Create_terrain.py script issues random crc16 warnings

Lazydude, 2020-05-28:
When using the create_terrain.py script, running on Ubuntu 20.04 with python 3.8.2, I randomly get the following two warnings:  

1. create_terrain.py:205: DeprecationWarning: PY_SSIZE_T_CLEAN will be required for '#' formats
  block.crc = crc16.crc16xmodem(buf)

2. create_terrain.py:223: DeprecationWarning: PY_SSIZE_T_CLEAN will be required for '#' formats
  crc2 = crc16.crc16xmodem(buf[:1821])

Is this expected?

Rand0409, 2022-08-29:
Hi,  I am using create_terrain.py and trying to upload it to drone by Mavftp, I am wondering have you fixed the issue?  And how you uploaded the DAT file by mavftp?

Thank you so much!

Rand0409, 2022-09-07:
I figured out and fixed the create_terrain issue... was on windows machine, that code does not support windows