ArduPilot/MissionPlanner #2397: MAVFtp issues random crc error which cancels ftp upload

Lazydude, 2020-05-28:
MAVFtp issues random crc error which cancels ftp upload

After downloadin map data using the create_terrain.py script and using MAVFtp to upload this map data to a Matek F765 running arduplane 4.06 beta I get this error:
Exception of type 'Ionic.ZipBadCrcException' was thrown.
at MissionPlanner.Controls.MavFTPUI.<>c_DisplayClass13_0.<UploadFile>b_2(IProgressReporterDialogueiprd)
at MissionPlanner.Controls.ProgressReporterDialogue.RunBackgroundOperation(Objecto)

This on Windows 10 using USB 2 and no hub.  It happens randomly and if I keep trying the upload eventually MAVFtp will upload the file with no issue.

Is this expected?

meee1, 2020-05-29:
how big where the files?

Lazydude, 2020-05-29:
All the files were either 1,856 KB or 1,792 KB.