ArduPilot/MissionPlanner #2400: Grid coordinates not accurate in some parts of the world

JonReacher, 2020-05-31:
#### Issue details
MGRS coordinates are not accurate in some parts of the world.

#### Version
MP 1.3.70

#### Platform
[ x ] All

I have been able to confirm that there are large areas of the globe that do not accurately give the correct MGRS coordinates. The Lat/Lon works just fine as far as I can tell everywhere. The same waypoint will be accurate in Lat/Lon but not in MGRS, but will be 10s of kilometers off, usually south.

meee1, 2020-05-31:
do you have a specific example?

JonReacher, 2020-05-31:
Yes, for instance if I place a waypoint on Entebbe airport in Uganda it gives me a coordinate of 36NVF3751900088. This coordinate is actually in Lake Victoria south of the airport.

JonReacher, 2020-05-31:
But here's the other thing I've noticed. If I create another waypoint anywhere and then manually enter that same coordinate it will actually go to the correct location. So I can have 2 waypoints with the exact same coordinate at different locations.

JonReacher, 2020-05-31:
Another example with larger disparity, clicked on Mbale town and it shows 36NXG3062309230. Create another waypoint and manually enter this and it goes to a (correct) location 10km due South. It seems to always be due South.

magicrub, 2020-06-01:
@JonReacher you're probably confusing lat/long units of decimal versus degrees in minutes and seconds. Common unit conversion mistake.

JonReacher, 2020-06-02:
@magicrub I'm not doing any conversion. Click on the flight plan map and it gives a wrong coordinate.

meee1, 2020-06-02:
@JonReacher can you please try the latest beta MP.

JonReacher, 2020-06-02:
@meee1 Beta appears to work perfectly. We may have to use lat/lon for now I guess, although not ideal, unless you have any suggestions. Would you not recommend using the beta for critical applications?

meee1, 2020-06-03:
at this point in time i would recommend using beta. there has been no large changes that i think would result in a bad outcome

closed 2020-06-08 by meee1