ArduPilot/MissionPlanner #2404: Extended tuning tab tuning value issue

Vissie26, 2020-06-03:
When I enter values for min/max under the "tune"  parameters. 
The scaling (if there is one) doesn't seem to be correct.
If i set the value at 0,191 for min and 0,250 for max. The values in the "full parameter list" are 191 and 250. 
[image]
Maybe I'm missing something but it doesn't seem right to me.

using latest full version of mission planner and arducopter 4.0.3

meee1, 2020-06-04:
can you goto setup > advanced > param gen, and then restart mp and test again. not sure if this is the issue, but worth a try

Vissie26, 2020-06-04:
Unfortunately it didn't solve the problem.

I later saw that this issue is also mentioned on the mission planner forum:
https://discuss.ardupilot.org/t/extended-tuning-possible-bug/56115