ArduPilot/MissionPlanner #2407: Write Waypoints - Immediate Crash

Ryanf55, 2020-06-07:
#### Issue details
If I plan a mission in MP and click "Write", it crashes immediately. 

#### Version
MP 1.3.71 build 1.3.7451.16917 APM:Copter V3.5.5 (27229c83)

#### Platform
[  ] All
[  ] AntennaTracker
[x  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
Hex

#### Hardware type
Pixhawk 1

#### Steps to reproduce

1. Start MP
1. Plug in autopilot
1. Connect (connection successful)
1. Click "Plan" tab
1. Click "Read" to get the stored mission
1. Click "Write" to write the mission

#### Logs
[image]

meee1, 2020-06-07:
Any reason you are still running 3.5.5

Ryanf55, 2020-06-08:
It was a known stable version for the functionality I was using and I haven't needed any of the new features. However, now I need AUTO missions, and this prevents me from creating them. I will upgrade APM FW and report back.

meee1, 2020-06-08:
i released a new version of MP yesterday,could you try that?

Ryanf55, 2021-04-21:
It worked. Forgot to update the ticket. Closing now.

closed 2021-04-21 by Ryanf55