ArduPilot/MissionPlanner #2410: HIGH_LATENCY messages that disappear from Script

expntly, 2020-06-09:
#### Issue details
While testing HIGH_LATENCY related comms, I noticed that every time a message is received by MP the HUD and map will properly reflect the latest telemetry, but the Python Script won't always see that HIGH_LATENCY message.  Sometimes it does, sometimes it doesn't.  If sending HIGH_LATENCY messages every 15 seconds to MP, there will be one or two misses per minute.

Script:

```python
while True:
  if not cs.connected:
    time.sleep(1)
    print("not connected yet")
    continue
  p = MAV.readPacket()
  print("msg: %s" % p.msgtypename)
```

What I see in the console is

```
msg:HEARTBEAT
msg:HEARTBEAT
msg:HEARTBEAT
msg:HIGH_LATENCY
msg:HEARTBEAT
...
```

And when there's a miss, I just see the HEARTBEAT messages scroll by but there's no HIGH_LATENCY printed in the console **even though** the HUD and map are properly updated.

It looks like MAV.readPacket() isn't forwarding all messages to the script?

#### Version
1.3.68.1

#### Platform
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

meee1, 2020-06-09:
using readpacket in a script creates a race condition. so MP is fighting with you script to read packets. its a 50/50 as to who gets it.

expntly, 2020-06-09:
What is the right way to read messages without messing with MP?  It feels like I'm doing the wrong thing if the script is not seeing all the packets and if it's stealing messages from MP.  That's the worst of both worlds.

That said, it doesn't really match what I'm seeing.  MP seems to be seeing 100% of the messages, but the script is seeing less than 100% of the messages.

meee1, 2020-06-10:
import clr
import MissionPlanner
clr.AddReference("MAVLink")
from System import Func
import MAVLink

def OtherMethod(message):
    print "got HB";
    print dir(message.data)
    return True

sub = MAV.SubscribeToPacketType(MAVLink.MAVLINK_MSG_ID.HEARTBEAT.value__, Func[MAVLink.MAVLinkMessage, bool] (OtherMethod))

#MAV.UnSubscribeToPacketType(MAVLink.MAVLINK_MSG_ID.HEARTBEAT.value__, sub);

meee1, 2020-06-10:
regarding altitude_sp 
you might be correct.
i think altasl is correct. but no alt relative is ever set, i am putting altitude_sp into that, which is wrong, as its the setpoint, not the current

meee1, 2020-06-10:
i will change it to this
                        altasl = highlatency.altitude_amsl;
                        alt = altasl - (float)HomeAlt;
                        alt_error = highlatency.altitude_sp - alt;

closed 2020-06-24 by meee1

expntly, 2020-07-05:
Thanks!  What is the best way to tell which release it's going into?

Related to the code you sent using the clr in python, I've tried to use it with STATUSTEXT instead of HEARTBEAT and it looks like it's not showing all status text messages.  For instance if I trigger a preflight calibration, my function gets called for the "Airspeed calibration started" msg, but the function does not get called for other messages that get sent by other mavlink nodes.  What's even stranger is that the `MAV.readPacket()` does see these messages sent by other nodes.

expntly, 2020-07-07:
I opened another issue with detailed information on this since this is not directly related to the HIGH_LATENCY issue.

https://github.com/ArduPilot/MissionPlanner/issues/2426

expntly, 2020-08-11:
@meee1 What is the best way to tell which MP release your June 9 "altasl fix" made it into?  Thanks

meee1, 2020-08-12:
its only in beta atm