ArduPilot/MissionPlanner #2412: Mission Planner stops with error on Windows 10 32bit

ronicontora, 2020-06-12:
#### Issue details
Mission Planner stops with error immediately after launch.

#### Version
Mission Planner 1.3.70 
Windows 10 Home 1511 32bit

#### Platform
[x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Hardware type
Tablet Samsung XE500T1C-A02RU

#### Logs
##### Windows Event log (Russian language pack is installed)
Приложение: MissionPlanner.exe
Версия платформы: v4.0.30319
Описание. Процесс был завершен из-за необработанного исключения.
Сведения об исключении: System.AccessViolationException
   в OpenTK.Platform.Windows.Functions.SwapBuffers(IntPtr)
   в OpenTK.Platform.Windows.WinGLContext.SwapBuffers()
   в OpenTK.Graphics.GraphicsContext.SwapBuffers()
   в OpenTK.GLControl.SwapBuffers()
   в MissionPlanner.Controls.HUD.OnPaint(System.Windows.Forms.PaintEventArgs)
   в MissionPlanner.Controls.HUD.Refresh()
   в MissionPlanner.Controls.HUD.OnResize(System.EventArgs)
   в System.Windows.Forms.Control.OnSizeChanged(System.EventArgs)
   в System.Windows.Forms.Control.UpdateBounds(Int32, Int32, Int32, Int32, Int32, Int32)
   в System.Windows.Forms.Control.UpdateBounds()
   в System.Windows.Forms.Control.WmWindowPosChanged(System.Windows.Forms.Message ByRef)
   в System.Windows.Forms.Control.WndProc(System.Windows.Forms.Message ByRef)
   в System.Windows.Forms.ScrollableControl.WndProc(System.Windows.Forms.Message ByRef)
   в System.Windows.Forms.ContainerControl.WndProc(System.Windows.Forms.Message ByRef)
   в System.Windows.Forms.UserControl.WndProc(System.Windows.Forms.Message ByRef)
   в System.Windows.Forms.Control+ControlNativeWindow.OnMessage(System.Windows.Forms.Message ByRef)
   в System.Windows.Forms.Control+ControlNativeWindow.WndProc(System.Windows.Forms.Message ByRef)
   в System.Windows.Forms.NativeWindow.Callback(IntPtr, Int32, IntPtr, IntPtr)
   в System.Windows.Forms.SafeNativeMethods.SetWindowPos(System.Runtime.InteropServices.HandleRef, System.Runtime.InteropServices.HandleRef, Int32, Int32, Int32, Int32, Int32)
   в System.Windows.Forms.Control.SetBoundsCore(Int32, Int32, Int32, Int32, System.Windows.Forms.BoundsSpecified)
   в System.Windows.Forms.Control.System.Windows.Forms.Layout.IArrangedElement.SetBounds(System.Drawing.Rectangle, System.Windows.Forms.BoundsSpecified)
   в System.Windows.Forms.Layout.DefaultLayout.xLayoutDockedControl(System.Windows.Forms.Layout.IArrangedElement, System.Drawing.Rectangle, Boolean, System.Drawing.Size ByRef, System.Drawing.Rectangle ByRef)
   в System.Windows.Forms.Layout.DefaultLayout.LayoutDockedControls(System.Windows.Forms.Layout.IArrangedElement, Boolean)
   в System.Windows.Forms.Layout.DefaultLayout.xLayout(System.Windows.Forms.Layout.IArrangedElement, Boolean, System.Drawing.Size ByRef)
   в System.Windows.Forms.Layout.DefaultLayout.LayoutCore(System.Windows.Forms.Layout.IArrangedElement, System.Windows.Forms.LayoutEventArgs)
   в System.Windows.Forms.Layout.LayoutEngine.Layout(System.Object, System.Windows.Forms.LayoutEventArgs)
   в System.Windows.Forms.Control.OnLayout(System.Windows.Forms.LayoutEventArgs)
   в System.Windows.Forms.ScrollableControl.OnLayout(System.Windows.Forms.LayoutEventArgs)
   в System.Windows.Forms.Control.PerformLayout(System.Windows.Forms.LayoutEventArgs)
   в System.Windows.Forms.Control.System.Windows.Forms.Layout.IArrangedElement.PerformLayout(System.Windows.Forms.Layout.IArrangedElement, System.String)
   в System.Windows.Forms.Control.OnResize(System.EventArgs)
   в System.Windows.Forms.Panel.OnResize(System.EventArgs)
   в System.Windows.Forms.Control.OnSizeChanged(System.EventArgs)
   в System.Windows.Forms.Control.UpdateBounds(Int32, Int32, Int32, Int32, Int32, Int32)
   в System.Windows.Forms.Control.UpdateBounds()
   в System.Windows.Forms.Control.WmWindowPosChanged(System.Windows.Forms.Message ByRef)
   в System.Windows.Forms.Control.WndProc(System.Windows.Forms.Message ByRef)
   в System.Windows.Forms.ScrollableControl.WndProc(System.Windows.Forms.Message ByRef)
   в System.Windows.Forms.Control+ControlNativeWindow.OnMessage(System.Windows.Forms.Message ByRef)
   в System.Windows.Forms.Control+ControlNativeWindow.WndProc(System.Windows.Forms.Message ByRef)
   в System.Windows.Forms.NativeWindow.Callback(IntPtr, Int32, IntPtr, IntPtr)
   в System.Windows.Forms.SafeNativeMethods.SetWindowPos(System.Runtime.InteropServices.HandleRef, System.Runtime.InteropServices.HandleRef, Int32, Int32, Int32, Int32, Int32)
   в System.Windows.Forms.Control.SetBoundsCore(Int32, Int32, Int32, Int32, System.Windows.Forms.BoundsSpecified)
   в System.Windows.Forms.Control.SetBounds(Int32, Int32, Int32, Int32, System.Windows.Forms.BoundsSpecified)
   в System.Windows.Forms.SplitterPanel.set_HeightInternal(Int32)
   в System.Windows.Forms.SplitContainer.set_SplitterDistance(Int32)
[... 362 lines cut ...]
   в System.Security.Cryptography.X509Certificates.PublicKey.get_Key()
2020-06-12 14:53:05,865 DEBUG Program - TypeResolve Failed: 2.5.29.35 from mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089 (:0) [TileLoader: 4]
2020-06-12 14:53:05,865 DEBUG Program - TypeResolve Failed: 1.3.6.1.5.5.7.1.1 from mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089 (:0) [TileLoader: 4]
2020-06-12 14:53:05,865 DEBUG Program - TypeResolve Failed: 2.5.29.17 from mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089 (:0) [TileLoader: 4]
2020-06-12 14:53:05,880 DEBUG Program - TypeResolve Failed: 2.5.29.32 from mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089 (:0) [TileLoader: 4]
2020-06-12 14:53:05,880 DEBUG Program - TypeResolve Failed: 2.5.29.31 from mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089 (:0) [TileLoader: 4]
2020-06-12 14:53:05,880 DEBUG Program - TypeResolve Failed: 1.3.6.1.4.1.11129.2.4.2 from mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089 (:0) [TileLoader: 4]
2020-06-12 14:53:05,927 DEBUG Program - FirstChanceException in: System (:0) [TileLoader: 0]
System.NotSupportedException: Алгоритм ключа сертификата не поддерживается.
   в System.Security.Cryptography.X509Certificates.PublicKey.get_Key()
2020-06-12 14:53:05,943 DEBUG Program - TypeResolve Failed: 2.5.29.35 from mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089 (:0) [TileLoader: 0]
2020-06-12 14:53:05,943 DEBUG Program - FirstChanceException in: System (:0) [TileLoader: 2]
System.NotSupportedException: Алгоритм ключа сертификата не поддерживается.
   в System.Security.Cryptography.X509Certificates.PublicKey.get_Key()
2020-06-12 14:53:05,943 DEBUG Program - TypeResolve Failed: 1.3.6.1.5.5.7.1.1 from mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089 (:0) [TileLoader: 0]
2020-06-12 14:53:05,943 DEBUG Program - TypeResolve Failed: 2.5.29.17 from mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089 (:0) [TileLoader: 0]
2020-06-12 14:53:05,943 DEBUG Program - TypeResolve Failed: 2.5.29.35 from mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089 (:0) [TileLoader: 2]
2020-06-12 14:53:05,943 DEBUG Program - TypeResolve Failed: 2.5.29.32 from mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089 (:0) [TileLoader: 0]
2020-06-12 14:53:05,959 DEBUG Program - TypeResolve Failed: 2.5.29.31 from mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089 (:0) [TileLoader: 0]
2020-06-12 14:53:05,959 DEBUG Program - TypeResolve Failed: 1.3.6.1.4.1.11129.2.4.2 from mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089 (:0) [TileLoader: 0]
2020-06-12 14:53:05,959 DEBUG Program - TypeResolve Failed: 1.3.6.1.5.5.7.1.1 from mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089 (:0) [TileLoader: 2]
2020-06-12 14:53:05,959 DEBUG Program - TypeResolve Failed: 2.5.29.17 from mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089 (:0) [TileLoader: 2]
2020-06-12 14:53:05,959 DEBUG Program - TypeResolve Failed: 2.5.29.32 from mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089 (:0) [TileLoader: 2]
2020-06-12 14:53:05,959 DEBUG Program - TypeResolve Failed: 2.5.29.31 from mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089 (:0) [TileLoader: 2]
2020-06-12 14:53:05,974 DEBUG Program - TypeResolve Failed: 1.3.6.1.4.1.11129.2.4.2 from mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089 (:0) [TileLoader: 2]

meee1, 2020-06-12:
update your graphics card driver.

meee1, 2020-06-24:
INFO MissionPlanner.Controls.HUD - Device: Intel(R) Graphics Media Accelerator 3600 Series
does not support opengl under windows 10

closed 2020-06-24 by meee1

ronicontora, 2020-06-24:
I downgrade to Windows 8.1 and Mission Planner run fine.